Azure/iotedge-lorawan-starterkit #442: 400 Bad Request on function app install with Azure Template

prettyflyforabeeguy, 2021-10-01:
Following Step 1 of the Step-By-Step Instructions 
https://github.com/Azure/iotedge-lorawan-starterkit/blob/master/Docs/quickstart.md
After the azuredeploy.json script has been running for a while, all resources appear to install correctly except for the function app.
I have tried deploying to an existing resource group and creating a new resource group
I've tried deploying to EAST and WEST US, and have attempted to redeploy.   It looks like someone reported this issue before here https://github.com/Azure/iotedge-lorawan-starterkit/issues/246

but unlike them, even after multiple attempts the error does not stop happening for me.  I have attempted more than 10 times in different regions and on different days.  Could something be wrong with the script ?

[image]

Mandur, 2021-10-05:
Hello @prettyflyforabeeguy 

Thank you very much for the issue report. I might have an idea. Would it be possible for you to try out the button on a custom branch to see if it fix it?

prettyflyforabeeguy, 2021-10-05:
I'll see if I can do that and report back.  Thanks!

prettyflyforabeeguy, 2021-10-05:
It looks like that worked! I installed Azure CLI, cloned the repo to your fix, authenticated, and ran az deployment group create --name testdeployment --resource-group IoTGateway --template-file azuredeploy.json from the templates folder and all appears to have been successful! 
[image]

closed 2021-10-07 by Mandur

prettyflyforabeeguy, 2021-10-12:
I'm having a separate issue now.  I'll email you a follow up.