Azure/iotedge-lorawan-starterkit #601: [bug] Unify .gitignore files

spygi, 2021-10-20:
## Current Behavior
- There are 5 different .gitignore files for different modules
- On the other hand there are 2 .editorconfigs (one for regular code and another for the tests)

## Expected behavior
Have a single .gitignore for the whole repo on the root of the repo.

atifaziz, 2021-10-21:
> * There are 5 different .gitignore files for different modules

BTW, I see 4 (based on 06348f116d9a18d566ecdeaa93b94ac08b174148 at time of this comment):

    $ git ls-files **/.gitignore
    LoRaEngine/.gitignore
    LoRaEngine/LoraKeysManagerFacade/.gitignore
    LoRaEngine/modules/LoRaWanNetworkSrvModule/.gitignore
    LoRaEngine/modules/LoRaWanPktFwdModule/.gitignore

spygi, 2021-10-21:
> BTW, I see 4 (based on [06348f1](https://github.com/Azure/iotedge-lorawan-starterkit/commit/06348f116d9a18d566ecdeaa93b94ac08b174148) at time of this comment):

Can be an off-by-1 error 😂