Azure/iotedge-lorawan-starterkit #800: `CloudToDeviceMessageSizeLimitShouldAbandonTests.Should_Abandon` test fails sometime

atifaziz, 2021-11-15:
## Expected Behavior

Expected unit tests to pass on each run.

## Current Behavior

Instead the `CloudToDeviceMessageSizeLimitShouldAbandonTests.Should_Abandon` (integration) test fails with the following details:

```
  Failed LoRaWan.Tests.Integration.CloudToDeviceMessageSizeLimitShouldAbandonTests.Should_Abandon(isConfirmed: False, hasMacInUpstream: True, hasMacInC2D: False, datr: "SF7BW125") [1 ms]
  Error Message:
   System.Net.NetworkInformation.NetworkInformationException : An error was encountered while querying information from the operating system.
---- System.AggregateException : One or more errors occurred. (Could not find a part of the path '/sys/class/net/veth8b44ff8/mtu'.)
-------- System.IO.DirectoryNotFoundException : Could not find a part of the path '/sys/class/net/veth8b44ff8/mtu'.
  Stack Trace:
     at System.Net.NetworkInformation.LinuxNetworkInterface.GetLinuxNetworkInterfaces()
   at LoRaWan.Tests.Common.Utility.GetMacAddress() in /home/runner/work/iotedge-lorawan-starterkit/iotedge-lorawan-starterkit/Tests/Common/Utility.cs:line 20
   at LoRaWan.Tests.Common.MessageProcessorTestBase..ctor() in /home/runner/work/iotedge-lorawan-starterkit/iotedge-lorawan-starterkit/Tests/Common/MessageProcessorTestBase.cs:line 51
   at LoRaWan.Tests.Integration.CloudToDeviceMessageSizeLimitBaseTests..ctor()
   at LoRaWan.Tests.Integration.CloudToDeviceMessageSizeLimitShouldAbandonTests..ctor()
   at System.RuntimeType.CreateInstanceDefaultCtor(Boolean publicOnly, Boolean skipCheckThis, Boolean fillCache, Boolean wrapExceptions)
----- Inner Stack Trace -----

----- Inner Stack Trace -----
   at Interop.ThrowExceptionForIoErrno(ErrorInfo errorInfo, String path, Boolean isDirectory, Func`2 errorRewriter)
   at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String path, OpenFlags flags, Int32 mode)
   at System.IO.FileStream.OpenHandle(FileMode mode, FileShare share, FileOptions options)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options)
   at System.IO.StreamReader.ValidateArgsAndOpenPath(String path, Encoding encoding, Int32 bufferSize)
   at System.IO.StreamReader..ctor(String path, Encoding encoding, Boolean detectEncodingFromByteOrderMarks)
   at System.IO.File.InternalReadAllText(String path, Encoding encoding)
   at System.IO.File.ReadAllText(String path)
   at System.Net.NetworkInformation.StringParsingHelpers.ParseRawIntFile(String filePath)
   at System.Net.NetworkInformation.LinuxIPv4InterfaceProperties.GetMtu()
   at System.Net.NetworkInformation.LinuxIPv4InterfaceProperties..ctor(LinuxNetworkInterface linuxNetworkInterface)
   at System.Net.NetworkInformation.LinuxIPInterfaceProperties..ctor(LinuxNetworkInterface lni)
   at System.Net.NetworkInformation.LinuxNetworkInterface..ctor(String name, Int32 index)
   at System.Net.NetworkInformation.LinuxNetworkInterface.GetOrCreate(Dictionary`2 interfaces, String name, Int32 index)
   at System.Net.NetworkInformation.LinuxNetworkInterface.<>c__DisplayClass5_0.<GetLinuxNetworkInterfaces>b__2(String name, LinkLayerAddressInfo* llAddr)
```

## Steps to Reproduce

The test failure is sporadic and [occurs during some CI builds](https://github.com/Azure/iotedge-lorawan-starterkit/runs/4190652447?check_suite_focus=true#step:11:567) and goes away with re-runs.

## Logs

- 📎 [`logs_3628.zip`](https://github.com/Azure/iotedge-lorawan-starterkit/files/7537042/logs_3628.zip)

Mandur, 2021-11-15:
@bastbu

bastbu, 2021-11-15:
Duplicate of #745. Let's track it there and close this issue

closed 2021-11-15 by bastbu