Azure/iotedge-lorawan-starterkit #827: `LnsProtocolMessageProcessorConcentratorDeduplicationIntegrationTests` test fails sometimes

atifaziz, 2021-11-17:
## Expected Behavior

Expected integration tests to pass on each run.

## Current Behavior

Instead the `LnsProtocolMessageProcessorConcentratorDeduplicationIntegrationTests.When_Same_Message_Comes_Multiple_Times_Result_Depends_On_Which_Concentrator_It_Was_Sent_From` test fails with the following details:

```
  Failed LoRaWan.Tests.Integration.LnsProtocolMessageProcessorConcentratorDeduplicationIntegrationTests.When_Same_Message_Comes_Multiple_Times_Result_Depends_On_Which_Concentrator_It_Was_Sent_From(isSameStation: False) [894 ms]
  Error Message:
   Assert.Equal() Failure
Expected: 1
Actual:   2
  Stack Trace:
     at LoRaWan.Tests.Integration.LnsProtocolMessageProcessorConcentratorDeduplicationIntegrationTests.When_Same_Message_Comes_Multiple_Times_Result_Depends_On_Which_Concentrator_It_Was_Sent_From(Boolean isSameStation) in /home/runner/work/iotedge-lorawan-starterkit/iotedge-lorawan-starterkit/Tests/Integration/LnsProtocolMessageProcessorConcentratorDeduplicationIntegrationTests.cs:line 109
--- End of stack trace from previous location where exception was thrown ---
```

## Steps to Reproduce

The test failure is sporadic and [occurs during some CI builds](https://github.com/Azure/iotedge-lorawan-starterkit/runs/4229340357?check_suite_focus=true#step:11:684) and goes away with re-runs.

## Logs

- 📎 [`logs_4021.zip`](https://github.com/Azure/iotedge-lorawan-starterkit/files/7552323/logs_4021.zip)

atifaziz, 2021-11-17:
/cc @spygi

spygi, 2021-11-17:
Thanks for reporting this, dup of #824

closed 2021-11-24 by spygi