Azure/iotedge-lorawan-starterkit #829: Logging statements execute a lot of work if log level is not enabled

bastbu, 2021-11-17:
## Expected Behavior

Expensive log statements, such as `logger.LogDebug($"cloud to device message: {((frmPayload?.Length ?? 0) == 0 ? "empty" : ConversionHelper.ByteArrayToString(frmPayload))}, id: {cloudToDeviceMessage.MessageId ?? "undefined"}, fport: {fport ?? 0}, confirmed: {requiresDeviceAcknowlegement}, cidType: {macCommandType}, macCommand: {macCommands.Count > 0}");` should only be executed if the log level is enabled.

## Current Behavior

Many expensive log statement operations (string interpolations, conversions, etc) are always executed.

Mandur, 2021-11-17:
@atifaziz looking if needed in .net 6

atifaziz, 2021-11-17:
Based on an initial look, it seems like that logging did not get any update with regards to optimised and optional string interpolation. Either we can add this support or go with a lightweight level-logging wrapper that short-circuits any work if the right logging level is not in effect. If we want to use structure logging instead, then there are other options still.