Azure/iotedge-lorawan-starterkit #1042: Cli-LoRa-Device-Provisioning should be provided as asset with release

danigian, 2021-12-10:


danigian, 2021-12-10:
@Mandur let's discuss in next sprint refinement

Mandur, 2022-10-25:
We will take this in the current sprint

ABHITHLALC, 2023-01-14:
Sorry , I don"t know if I can place this doubt here. I would be very grateful if you could help. 
I'm new to Azure IoT Edge LoRaWAN Starter Kit and following [https://azure.github.io/iotedge-lorawan-starterkit/dev/](url)
I'm having trouble to understanding how to find CLI-LoRa-Device-Provisioning interface. Is it in our lora edge gateway or in azure iot  hub platform or in our machine? Could please give me some guidence.

Mandur, 2023-01-14:
Hello @ABHITHLALC 

You can find the CLI compiled in every release (e.g. [here ](https://github.com/Azure/iotedge-lorawan-starterkit/releases/tag/v2.2.1))

You can find the code backing it [here](https://github.com/Azure/iotedge-lorawan-starterkit/tree/dev/Tools/Cli-LoRa-Device-Provisioning)

ABHITHLALC, 2023-01-16:
Hi @Mandur,
Thank you for the reply. And sorry for the delay.
Still having a doubt. Probabaly this will be low level questions. But still asking ,where do we run this commands?
```
dotnet publish -c Release -r win10-x64
dotnet publish -c Release -r linux-x64
dotnet publish -c Release -r osx-x64
dotnet run -- (add verbs and parameters here)

```
I am using SX1302 LoRaWAN Gateway HAT and raspberry pi 3 B+ . Steps followed
1. Installed sd card with debian OS
2. connected sx1302 hat with raspberry pi.
3. Installed Iotedge in raspberry pi
4. In azure iot hub , deployed lorawan starter kit  from this document [https://azure.github.io/iotedge-lorawan-starterkit/dev/quickstart/#deployed-azure-infrastructure](url)
5.Copied connection string obtained after deploying  starter kit.
6. And used same connection string to provision the edge gateway
7. Now the edge modules are running 
[image]

But after this I can see from the logs of LoRaWanNetworkServerModule, There is an error saying 
```
An exception occurred while processing requests: LoRaWan.LoRaProcessingException: The configuration request of station 'B827EBFFFEA63045' did not match any configuration in IoT Hub. If you expect this connection request to succeed, make sure to provision the Basics Station in the device registry.
ErrorCode: InvalidDeviceConfiguration
```

[image]

Could you please provide me with more inofrmation about LBS IoT Edge Module configuration. Your guidence would be very helpful.

Mandur, 2023-01-17:
Hello @ABHITHLALC,

You are missing a device for the concentrator with id 'B827EBFFFEA63045'. YOu don't need to compile the utility, as my previous post suggested, just go [here](https://github.com/Azure/iotedge-lorawan-starterkit/releases/tag/v2.2.1) in the assets section and download the lora-cli.your-platform.zip, unzip it, and run the executable, this is a precompiled version of the utility for you

If you prefer you could also put the configuration manually as described [here](https://azure.github.io/iotedge-lorawan-starterkit/2.2.1/user-guide/station-device-provisioning/#manual-configuration)

ABHITHLALC, 2023-01-18:
Thanks,  @Mandur 
This ID(B827EBFFFEA63045) is not created by me. I think that generated automatically after installation. 
deployed another iotedgelorawan starter kit with station eui ' **B827EBFFFEA63045** '. Now the logs are
[IotHubMetricsCollectorModule logs.txt](https://github.com/Azure/iotedge-lorawan-starterkit/files/10442074/IotHubMetricsCollectorModule.logs.txt)
[LoRaWanNetworkSrvModule logs.txt](https://github.com/Azure/iotedge-lorawan-starterkit/files/10442075/LoRaWanNetworkSrvModule.logs.txt)
[iotedge system logs.txt](https://github.com/Azure/iotedge-lorawan-starterkit/files/10442062/iotedge.system.logs.txt)
Saying that router config was not present in device tiwn.

As you said I downloaded the file from [here](https://github.com/Azure/iotedge-lorawan-starterkit/releases/tag/v2.2.1). I am using system with windows 10. So downloaded " **lora-cli.win-x64.zip** ". That file contains  these items
[image]
Ran loradeviceprovisioning.exe in CMD. Then suddenly a window pops up and closes. Am I in the right path?