Azure/iotedge-lorawan-starterkit #1108: FSK messages are not being processed by Basic Station, even though configuration seems apparently ok

danigian, 2021-12-16:
Even though we are sending to Basic Station the configuration with a DR7 set to [0,0,0] (as required for FSK) and this seems to be apparently processed by station (as per image below), when sending a message with the "red device" we can't receive it on the basic station side.

We should investigate if the behavior is due to a misconfiguration on our side or to something else
[image]

danigian, 2021-12-16:
@Mandur, @atifaziz for your info

roel4ez, 2022-01-05:
ping @mandur: do we need to tackle this for the January release?

Mandur, 2022-01-05:
No it is out of score for January