Azure/iotedge-lorawan-starterkit #1120: Mac Commands piggybacked on the payload should not be ignored

Mandur, 2021-12-17:
[image]
Currently we only process mac commands in case they are in the main payload. We also need to process the mac commands piggybacked in the Fopts

p-schuler, 2022-01-03:
@danigian - I think there are 2 issues here: the fopts support would be additional support (new feature), the key mismatch would be a bug. I wonder how that would work though? 

fyi @Mandur

danigian, 2022-01-05:
I will tackle both issues in the same PR:
- Making sure that Mac Commands in FOpts are not ignored
- Making sure that when a FPort 0 is used, payload is decrypted with NwkSKey

danigian, 2022-01-07:
I created #1203 for tracking the decryption issue

PR #1204 (2022-01-11): Proper Mac Commands handling
# Mac Commands handling

## What is being addressed

This PR is tackling following issues:

- [x] Fixes #1203
- [x] Fixes #1120
- [x] Fixes #1174
 
## How is this addressed

- [x] Logic in DefaultLoRaDataRequestHandler has been modified 
- [x] When_Mac_Command_Should_Not_Send_Upstream_Messages_And_Should_Skip_Calls_To_FrameCounterDown_When_SoftDuplicate integration test has been re-enabled for properly working
- [x] Persists_Reported_Dwell_Time_On_TxParamSetupAns DwellTimeIntegrationTest has been modified to test both FOpts and Payload sending of the Mac Command An [... cut]
closed 2022-01-11 by danigian