Azure/iotedge-lorawan-starterkit #1125: Rctx commands should be transmitted as is to downstream message

Mandur, 2021-12-20:
We are using the RCTX value coming from a basicsstation to derive an antenna preference when building a dowstream message. This is wrong, according to the NS protocol we should simply reply with the value received

danigian, 2022-01-03:
Rctx is parsed from uplink messages and put in the AntennaPreference field of the RadioMetadataUpInfo class.
No one is changing the AntennaPreference field and no computation is done on that field.
Therefore, according to the LNS protocol we are replying with the value received.

Closing the issue, but feel free to reopen it @Mandur if I misunderstood your point

Mandur, 2022-01-03:
Ok I understand better, AntennaPreference == RCTX right? I missunderstood the field and thought there was a logic behind it (mixed with another field from the pkt fwd).  what about naming the field RCTX or at least adding a comment to avoid missconfiguration?

Anyway we should then uncomment the tests that I wrongly commented out:

https://github.com/Azure/iotedge-lorawan-starterkit/blob/3569ed567ff3bae3bc24bfad55c4b699ee485fc5/Tests/Unit/NetworkServer/BasicsStation/DownstreamSenderTests.cs#L91

https://github.com/Azure/iotedge-lorawan-starterkit/blob/3569ed567ff3bae3bc24bfad55c4b699ee485fc5/Tests/Unit/NetworkServer/BasicsStation/DownstreamSenderTests.cs#L148

danigian, 2022-01-03:
I have decided to not rename the RCTX field as I think that AntennaPreference is more vocal than RCTX 👍 Added a comment though to avoid misconfiguration.

Also reverted the changes you did for tests in PR #1168

closed 2022-01-04 by danigian