Azure/iotedge-lorawan-starterkit #1131: Observability & Logs on ELK

ijardillier, 2021-12-20:
We are using a complete Elastic stack to monitor our projects based on this IoT Edge LoraWan project.

It uses Prometheus to expose metrics and it is fully compatible with the Elastic stack (metricbeat module is able to read prometheus or openmetrics endpoints).

But for logs, is it possible to provide structured logs (as Json objetcs). To do this, it is possible for example to use NLog or Serilog loggers (just a library in top of Microsoft.Extensions.Logging, so it can use the ILogger interface as well).

Elastic provides [Nugets](https://github.com/elastic/ecs-dotnet) to format log messages as Json objects and made them compliant with the [(ECS)](https://www.elastic.co/guide/en/ecs/current/ecs-log.html).

For example, with Serilog, we can easily use the ECS formatter in the appsettings Serilog section :

`"WriteTo": [
      {
        "Name": "Console",
        "Args": {
          "formatter": "Elastic.CommonSchema.Serilog.EcsTextFormatter, Elastic.CommonSchema.Serilog"
        }
      }
    ]`

and quite the same for NLog in the NLog section: 

`"extensions": [
      { "assembly": "NLog.Extensions.Logging" },
      { "assembly": "NLog.Web.AspNetCore" },
      { "assembly": "NLog.Targets.ElasticSearch" },
      { "assembly": "Elastic.CommonSchema.NLog" }
    ],
  "targets": {
      "console": {
        "type": "Console",
        "layout": {
          "type": "EcsLayout"
        }
      }`

Metal-Mighty, 2022-04-21:
Hi all! 
Adding to this enhancement by @ijardillier, if it gets implemented it would be very nice to also include context information to the published logs. 
One useful context information we would like would be the message identifier (CorrelationId ?) of the message received by the LNS from the LBS in order to track the message in the different application layers. This ID is already provided when forwarding messages to a different module with a route.

bastbu, 2022-04-25:
Hello @ijardillier and @Metal-Mighty and thanks for your inputs! We were considering adding this feature, sadly it's not as straightforward as pulling in a NuGet package (yet). Since we are using `Microsoft.Extensions.Logging` we would need a different library (i.e. [ecs-dotnet](https://github.com/elastic/ecs-dotnet/blob/main/src/Elasticsearch.Extensions.Logging/ReadMe.md)) to support this. Their repository has an open [feature request](https://github.com/elastic/ecs-dotnet/issues/125) to support the ECS format with `Microsoft.Extensions.Logging`, which we would like to reuse.

cc @Mandur

Metal-Mighty, 2023-03-29:
Hi @Mandur and thanks for your answer! I may have overlooked the last part of the message 😶
I'll try to follow up with them to see whether they can implement it.

In the mean time I added a new issue (#2069) to at least have a timestamp in console logs to help us during debug when we can't use our ELK logs.