Azure/iotedge-lorawan-starterkit #1137: Configuration from IoT device should be properly validated.

Mandur, 2021-12-21:
Missconfiguration on the iot Hub devices should be handled at parsing time. logging when it is not a breaking exception, or stopping the device creation.

p-schuler, 2022-01-03:
@Mandur - can we add more details here? Like what parsing in particular, which values? We can then consider it for the current milestone.

Mandur, 2022-01-03:
Boy I thought we could easily relate back to the PR discussion from here. Here is the direct link https://github.com/Azure/iotedge-lorawan-starterkit/pull/1122#discussion_r772482857

Mandur, 2022-01-03:
[image]
As it seems it still doesn't work here is a screenshot of the discussion