Azure/iotedge-lorawan-starterkit #1139: LnsRXDelay should not depend on receiveWindow

Mandur, 2021-12-21:
In the downlinkmessagebuilder we have logic where we make the lnsrxdelay dependant of the rxwindows we are selecting (either because of timing or of twins property). The lnsrxdelay in both cases should always be 1 (+ eventual RXdelay coming from the lora devices), if we want to force a device to use second windows (= preferredwindows == 2), we should omit the RX1DR and RX1Freq from the downstream message as we do for class C.

danigian, 2022-01-03:
Waiting for #1143 to be merged before going on with this

Mandur, 2022-01-09:
Agreeed. My understanding of this evolved substantially since I wrote this, I hope you synced with @atifaziz. I am sorry I should have commented here. Additionally, I think we had a missunderstanding on the RxDelay setting in the initial implementaiton, here is my current thinking, let me know if you agree:

* The ReportedRxDelay setting should be defaulted to 1 (instead of now 0) for normal devices, in case of class C, we should default it to 0. Logic should be adapted if that causes issues (I don't see any major pitfall, but...)
* LnsRxDelay should be deleted and everyone should use the value in the ReportedRxDelay

danigian, 2022-01-10:
As discussed, let's keep this bug open for future improvements.
I will tackle in #1213 the fact that if we want to force a device to use second window (= preferredwindows == 2), we should omit the RX1DR and RX1Freq from the downstream message sent to basic station.

danigian, 2022-01-31:
This was fixed in #1179

closed 2022-01-31 by danigian