Azure/iotedge-lorawan-starterkit #1169: Duplicate (identical) models between `LoraKeysManagerFacade` and `LoRaWan.NetworkServer`

bastbu, 2022-01-03:
## Expected Behavior

Identical types should only occur once in the solution. If these types are reused across boundaries that (de-)serialize them from JSON, reusing the same type will also give stronger guarantees that the contract between Network Server and Facade is not broken (currently some of these contracts are untested, which can lead to bugs such as #1149).

## Current Behavior

Some (identical) models that should be shared between the facade and the networkserver are actually duplicated (e.g. `FunctionBundlerRequest`, `FunctionBundlerResult`, etc). Some other models share only some of the properties (e.g. `IoTHubDeviceInfo`).

roel4ez, 2022-01-04:
check with @mandur why this has been separated

Mandur, 2022-01-09:
I have no idea why it was built like this, if the classes are similar I think this should be harmonized