Azure/iotedge-lorawan-starterkit #1230: Data_Rate_Is_Updated_When_C2D_With_LinkADRCmd_Received is frequently failing

danigian, 2022-01-11:
E2E test Data_Rate_Is_Updated_When_C2D_With_LinkADRCmd_Received is frequently failing.

From logs, apparently either the C2D message does not get sent or it's not processed by our LNS.

We should investigate

MaggieSalak, 2022-01-12:
I reproduced the issue locally where it appears that most of the time the C2D message is never arriving in the LNS. Tried adding  `TestFixtureCi.CleanupC2DDeviceQueueAsync()` before sending the C2D message and using the Function endpoint `SendCloudToDeviceMessage` instead of the service client but those options didn't solve the problem.

MaggieSalak, 2022-01-12:
When running the E2E test locally with the LNS also running locally, I was able to see the C2D message received and handled correctly by after that the subsequent messages are no longer logged at all in the LNS, therefor the condition that the test asserts is never met:

FF00000000000032 cloud to device MAC command LinkADRCmd received Type: LinkADRCmd Answer, datarate: 3, txpower: 4, nbTrans: 1, channel Mask Control: 0, Redundancy: 1
FF00000000000032 cloud to device message: empty, id: undefined, fport: 0, confirmed: False, cidType: Zero, macCommand: True
FF00000000000032 using standard region RX2 datarate DR0
FF00000000000032 using standard region RX2 frequency 869525000
FF00000000000032 UnconfirmedDataDown {"DevEui":{"AsUInt64":18374686479671623730},"LnsRxDelay":2,"Xtime":51228446143913539,"AntennaPreference":0,"DataRateRx1":0,"DataRateRx2":0,"FrequencyRx1":{"AsUInt64":869525000,"InKilo":869525.0,"InMega":869.525,"InGiga":0.869525},"FrequencyRx2":{"AsUInt64":869525000,"InKilo":869525.0,"InMega":869.525,"InGiga":0.869525},"Data":{"Length":18,"IsEmpty":false},"StationEui":{"AsUInt64":13269834311794917815}}
FF00000000000032 UnconfirmedDataDown {"DevEui":{"AsUInt64":18374686479671623730},"LnsRxDelay":2,"Xtime":51228446143913539,"AntennaPreference":0,"DataRateRx1":0,"DataRateRx2":0,"FrequencyRx1":{"AsUInt64":869525000,"InKilo":869525.0,"InMega":869.525,"InGiga":0.869525},"FrequencyRx2":{"AsUInt64":869525000,"InKilo":869525.0,"InMega":869.525,"InGiga":0.869525},"Data":{"Length":18,"IsEmpty":false},"StationEui":{"AsUInt64":13269834311794917815}}
FF00000000000032 completing cloud to device message, id: undefined
FF00000000000032 sending message to station with EUI 'B8-27-EB-FF-FE-D3-81-B7' with diid -1763149927. Payload '603200000380010000CBD4FCB2AB86F86A26'.

MaggieSalak, 2022-01-18:
I looked into this today and couldn't reproduce the issue anymore. In the E2E CI this test failed for the last time 1 week ago https://github.com/Azure/iotedge-lorawan-starterkit/runs/4780913652?check_suite_focus=true. It's possible that long timeouts that caused the request queue for the device to be blocked for a longer time than necessary (described in https://github.com/Azure/iotedge-lorawan-starterkit/issues/1190) was causing these failures. I would suggest closing this bug and reopening if we see this particular E2E test failing again.

closed 2022-01-18 by MaggieSalak