Azure/iotedge-lorawan-starterkit #1262: In some cases, ItestArm1 Network Server on CI seems to crash without trace

Mandur, 2022-01-13:
## Expected Behavior

The Network Server should never crash

## Current Behavior

In the CI we are seeing some occurrence where we see the Basicsstation reconnecting to the network server. This seems to be triggered by a Network server restart. We were able to run docker logs -f during one of such even and the network server logs got interrupted whereas the basicsstation. Here are the logs from the occurence:

BasicsStation:
```
2022-01-11 08:17:14.607 [S2E:DEBU] 1f5:a9ff:0:5 diid=-1436446425 [ant#0] - next TX start ahead by 366ms767us
2022-01-11 08:17:14.673 [SYN:VERB] Time sync rejected: quality=150 threshold=100
2022-01-11 08:17:14.954 [S2E:VERB] 1f5:a9ff:0:5 diid=-1436446425 [ant#0] - starting TX in 19ms905us
2022-01-11 08:17:14.979 [any:XDEB] DC EU band 1000 blocked until 21:19:03.220 (txtime=21:16:38.836 airtime=144ms384us)
2022-01-11 08:17:14.979 [any:XDEB] DC dnchnl 1 blocked until 21:16:44.034 (txtime=21:16:38.836 airtime=144ms384us)
2022-01-11 08:17:14.979 [AIO:XDEB] [3|WS] > {"msgtype":"dntxed","seqno":-1436446425,"diid":-1436446425,"DR":3,"Freq":867700000,"DevEui":"01-F5-A9-FF-00-00-00-05","rctx":0,"xtime":19421778249294572,"txtime":1199798.836722,"gpstime":0}
2022-01-11 08:17:14.979 [S2E:INFO] TX 1f5:a9ff:0:5 diid=-1436446425 [ant#0] - dntxed: 867.7MHz 16.0dBm ant#0(0) DR3 SF9/BW125 frame=60B0B12803A00100245E9412
2022-01-11 08:17:15.118 [S2E:DEBU] Tx done diid=-1436446425
2022-01-11 08:17:16.773 [SYN:VERB] Time sync rejected: quality=117 threshold=100
2022-01-11 08:17:25.256 [AIO:DEBU] [3] Connection closed unexpectedly
2022-01-11 08:17:25.256 [AIO:DEBU] [3] WS connection shutdown...
2022-01-11 08:17:25.257 [TCE:VERB] Connection to MUXS closed in state 4
``` 

Network Server:
``` 
01F5A9FF00000005 message '{"value":108}' sent to hub
01F5A9FF00000005 checking cloud to device message for 00:00:00.4388175
01F5A9FF00000005 done checking cloud to device message, found no message
01F5A9FF00000005 using standard region RX2 datarate DR0
01F5A9FF00000005 using standard region RX2 frequency 869525000
01F5A9FF00000005 UnconfirmedDataDown {"DevEui":{"AsUInt64":141205876013793285},"LnsRxDelay":1,"Xtime":19421778248294572,"AntennaPreference":0,"DataRateRx1":3,"DataRateRx2":0,"FrequencyRx1":{"AsUInt64":867700000,"InKilo":867700.0,"InMega":867.7,"InGiga":0.8677},"FrequencyRx2":{"AsUInt64":869525000,"InKilo":869525.0,"InMega":869.525,"InGiga":0.869525},"Data":{"Length":12,"IsEmpty":false},"StationEui":{"AsUInt64":13269834311785941743}}
01F5A9FF00000005 UnconfirmedDataDown {"DevEui":{"AsUInt64":141205876013793285},"LnsRxDelay":1,"Xtime":19421778248294572,"AntennaPreference":0,"DataRateRx1":3,"DataRateRx2":0,"FrequencyRx1":{"AsUInt64":867700000,"InKilo":867700.0,"InMega":867.7,"InGiga":0.8677},"FrequencyRx2":{"AsUInt64":869525000,"InKilo":869525.0,"InMega":869.525,"InGiga":0.869525},"Data":{"Length":12,"IsEmpty":false},"StationEui":{"AsUInt64":13269834311785941743}}
01F5A9FF00000005 sending message to station with EUI 'B8-27-EB-FF-FE-4A-8A-EF' with diid -1436446425. Payload '60B0B12803A00100245E9412'.
01F5A9FF00000005 converted 16bit FCnt 5 to 32bit FCnt 5
01F5A9FF00000005 processing time: 00:00:00.6111078
01F5A9FF00000005 Data message received from station 02-42-A3-FF-FE-BF-2F-A2. Marked as SoftDuplicateDueToDeduplicationStrategy because message already encountered.
01F5A9FF00000005 resubmit from confirmed message detected, msg: 5 server: 5
01F5A9FF00000005 down frame counter: 1
01F5A9FF00000005 decoding with: DecoderValueSensor port: 8
01F5A9FF00000005 sending message {"time":2814750437018316,"tmms":0,"freq":867.7,"chan":3,"rfch":0,"modu":"LoRa","datr":"SF9BW125","rssi":-53.0,"lsnr":12.0,"data":{"value":108},"port":8,"fcnt":5,"edgets":1641889034026,"rawdata":"MTA4","eui":"01F5A9FF00000005","gatewayid":"itestarm1","dupmsg":true} to hub
01F5A9FF00000005 message '{"value":108}' sent to hub
01F5A9FF00000005 skipping downstream messages due to deduplication (00:00:00.6575233)
01F5A9FF00000005 processing time: 00:00:00.6593630
B827EBFFFE4A8AEF Received 'dntxed' message: '{"msgtype":"dntxed","seqno":-1436446425,"diid":-1436446425,"DR":3,"Freq":867700000,"DevEui":"01-F5-A9-FF-00-00-00-05","rctx":0,"xtime":19421778249294572,"txtime":1199798.836722,"gpstime":0}'
```

Another thing we noted is that the swap memory starts to be very full after such events:

Itestarm1 (where the problem occurs:)
[image]

Itestarm2:
[image]

danigian, 2022-01-19:
I think the issue is not a "crash" but rather a failure in detecting when the edge gateway is actually applying the changes in configuration and restarting the network server module.

As we can see from following [run](https://github.com/Azure/iotedge-lorawan-starterkit/runs/4857135784?check_suite_focus=true)
[image]
the tests were already running when the new configuration was instead grabbed and applied
[image]

roel4ez, 2022-01-19:
- [ ] Don't start a testrun before the modules are restarted

roel4ez, 2022-01-20:
see #1326