Azure/iotedge-lorawan-starterkit #1282: LNS does not handle `FPort` = 1 for data frames (`updf`)

atifaziz, 2022-01-14:
## Expected Behavior

Expected LNS to handle data frames where `FPort` is -1 (meaning missing), as per the spec:

    {
    "msgtype"   : "updf",
    "MHdr"      : UINT,
    "DevAddr"   : INT32,
    "FCtrl"     : UINT,
    "FCnt",     : UINT,
    "FOpts"     : "HEX",
    "FPort"     : INT(-1..255),
    "FRMPayload": "HEX",
    "MIC"       : INT32,
    ..
    }

## Current Behavior

The LNS rejects the message as being invalid with the message: _Received unexpected 'jreq' message: …."_

The problem seems to stem from the fact that the `Port` is read as an unsigned byte (see line 98):

https://github.com/Azure/iotedge-lorawan-starterkit/blob/f19bd7b5d8c060d77d9a30f938276de561f2cbca/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/JsonHandlers/LnsData.cs#L91-L106

danigian, 2022-01-14:
As additional detail, this scenario might only happen if FRMPayload is empty. FPort MUST be specified if FRMPayload is filled instead
[image]

atifaziz, 2022-01-14:
@danigian Right, so either:

- `FPort` is -1 and `FRMPayload` an empty string
- `FPort` is a byte (0…255) and `FRMPayload` a non-empty string whose length is always even (hexadecimal encoding so always two characters per byte)

closed 2022-01-27 by atifaziz