Azure/iotedge-lorawan-starterkit #1310: deployment.template.json references module sensordecodermodule not included in source

ksaye, 2022-01-18:
## Expected Behavior
When deploying the deployment.template.json, it references a sensordecodermodule that is not part of the modules and the deployment shows an error in EdgeAgent log files

## Current Behavior
either include the sensordecodermodule code or remove the module from the template manifest.  This is shown here: https://github.com/Azure/iotedge-lorawan-starterkit/blob/dev/LoRaEngine/deployment.template.json#L147

## Steps to Reproduce

1. clone the repo
2. create the .env file with the correct Contaier_Registry information
3.right click on deployment.template.json and select "Generate IoT Edge Deployment Manifest"
4. open the deployment.json and note the added module sensordecodermodule which is not included in the source

## Context (Environment)

### Device (Host) Operating System
Ubuntu 20.04

### Architecture

AMD64

### Container Operating System

Linux Containers

### LoRaWAN Module Version

#### Docker

## Logs

## Additional Information

roel4ez, 2022-01-19:
@ksaye we'll investigate in the current sprint and keep you informed. Thanks for reporting!

danigian, 2022-01-19:
Thanks @ksaye for reporting this. We have now removed the module from the template.

closed 2022-01-19 by danigian