Azure/iotedge-lorawan-starterkit #1311: deployment.template.json references variables that are not defined

ksaye, 2022-01-18:
## Expected Behavior
Generate a valid deployment.json.

## Current Behavior
When you generate the deployment.json, there are several variables that do not resolve and they cause the deployment to fail.  These include:

$EDGE_AGENT_VERSION
$EDGE_HUB_VERSION
$EDGEHUB_OPTIMIZEFORPERFORMANCE
$EDGEHUB_MQTTSETTINGS_ENABLED

```
          "edgeAgent": {
            "type": "docker",
            "settings": {
              "image": "mcr.microsoft.com/azureiotedge-agent:$EDGE_AGENT_VERSION",
              "createOptions": ""
            }
          },
          "edgeHub": {
            "type": "docker",
            "status": "running",
            "restartPolicy": "always",
            "settings": {
              "image": "mcr.microsoft.com/azureiotedge-hub:$EDGE_HUB_VERSION",
              "createOptions": "{\"HostConfig\":{\"PortBindings\":{\"8883/tcp\":[{\"HostPort\":\"8883\"}],\"443/tcp\":[{\"HostPort\":\"443\"}]}}}"
            },
            "env": {
              "OptimizeForPerformance": {
                "value": "$EDGEHUB_OPTIMIZEFORPERFORMANCE"
              },
              "mqttSettings__enabled": {
                "value": "$EDGEHUB_MQTTSETTINGS_ENABLED"
              },
              "AuthenticationMode": {
                "value": "CloudAndScope"
              }
            }
          }
```

## Steps to Reproduce

1. clone the repo
2. populate the .env file
3. generate the manifest

## Context (Environment)

### Device (Host) Operating System

Ubuntu 20.04

### Architecture

AMD64

### Container Operating System

Linux

### LoRaWAN Module Version

#### Docker

## Logs

## Additional Information

roel4ez, 2022-01-19:
@mandur do you have any feedback here?

@ksaye we will investigate in the current sprint and keep you informed. Thanks for reporting!

Mandur, 2022-01-19:
Hello @ksaye 

Thank you for reporting the issue!

We indeed added some more variables to give more tailoring options and we forgot to surface a section in the example.env. Could you please try to add the [green section here ](https://github.com/Azure/iotedge-lorawan-starterkit/commit/61030d0f2882c52795727c118a0659ab9dbcd093)to your .env file to see if it resolves your issue?

If it does we will merge this with dev.

ksaye, 2022-01-19:
@Mandur that section will absolutely fix the issue, I was just trying to capture the issue as the documentation is now broken and if a beginner tries to deploy the current code following the document, it will fail.

Mandur, 2022-01-19:
Great, I created #1311 to include the fix in dev. Thank you very much for reporting this @ksaye, let us know if you face any other issue!

closed 2022-01-21 by Mandur