Azure/iotedge-lorawan-starterkit #1313: Device Join error - "failed to handle join request. Value cannot be null. (Parameter 'upstreamDataRate')" using 2.0.0 beta bits

gerfen, 2022-01-18:
## Expected Behavior

I'm in the process of testing the new 2.0.0 beta bits.  I have set up a network server on a Linux VM and I have configured a Milesite LG67 gateway as a basic station.  The basic station successfully connects to the network server.  I have added a LoRaWAN device to Azure Iot Hub.  When this device tries to join the network I see the error in the logs (shown below).

It feels like I'm missing some configuration for the basic station device.

## Current Behavior

The device should successfully join the network.

### Device (Host) Operating System

Network Server: Ubuntu 20.04
Basic Station:  Open WRT

### Architecture

AMD64

### Container Operating System

Linux containers

## Logs

```
24E124136A391163 join request received
24E124136A391163 querying the registry for OTAA device
24E124136A391163 using iotHub directly, no edgeHub queue
24E124136A391163 getting device twin
24E124136A391163 done getting device twin
24E124136A391163 Device registered in cache.
24E124136A391163 updating twin
24E124136A391163 twin updated
24E124136A391163 the provided RX2DataRate is not valid
24E124136A391163 Device registered in cache.
24E124136A391163 failed to handle join request. Value cannot be null. (Parameter 'upstreamDataRate')
```

## Additional Information

@Mandur assisted me to get the network server and basic station configured.  At the time I did not have the AppKey and AppEUI for the LoRaWAN device so we were not able to complete all of the configuration

roel4ez, 2022-01-19:
Hi @gerfen, we will have a look at this in the current sprint and keep you informed! Thanks for reporting.

p-schuler, 2022-01-19:
@bastbu @MaggieSalak - looks like we have added stronger enforcement on parameters in US915 with 84dfed672f4fa0728634abab13ee97d766fea89c. TryGetDownstreamChannelFrequency does throw if upstreamDataRate is not specified, which is optional on the interface, hence the compiler does not catch it. We need to pass the parameter from all the callers (other paths beside the JoinRequestMessageHandler are affected). Let's make sure we have all those enforcements validated.

gerfen, 2022-01-19:
@roel4ez  Thanks for looking into the issue.  I'll have access to the Milesite gateway until the end of the week - Milesite has kindly provided remote access to a gateway so I can evaluate it as a suitable option for my customers.  If a new container module can be produced before then I can test the fix.  If not, I will get another gateway that I have in my possession configured as a basics station but I will need access to the basics station container module.

bastbu, 2022-01-20:
@gerfen The bug was fixed with #1320, hence on the latest version of `dev` you should not see the join errors any longer. We would appreciate it if you could test the latest version by creating and testing an updated (private) version of the container module using one of the Dockerfiles at `LoRaEngine\modules\LoRaWanNetworkSrvModule`.

gerfen, 2022-01-20:
@bastbu Sure.  I've built and deployed the network server module to my VM.  Unfortunately the gateway I'm testing as the basics station is currently offline and I'll need to wait until someone from Milesite can get it back online sometime later this afternoon my time.  I'll report back after I've had a chance to test.

gerfen, 2022-01-21:
I was able to verify that the issue is fixed. 

I did not catch the logs for the join but I can show the logs for a message being decoded and transmitted.

```24E124136A391163 converted 16bit FCnt 98 to 32bit FCnt 98
24E124136A391163 no Deduplication Strategy selected
24E124136A391163 FunctionBundler request finished preparing.
24E124136A391163 FunctionBundler request finished preparing.
24E124136A391163 FunctionBundler request: {"GatewayId":"azure-lora-networkserver","ClientFCntUp":98,"ClientFCntDown":101,"Rssi":-46,"AdrRequest":{"DataRate":0,"RequiredSnr":-15,"PerformADRCalculation":false,"FCntUp":98,"FCntDown":101,"MinTxPowerIndex":13,"GatewayId":"azure-lora-networkserver","ClearCache":false},"FunctionItems":5}
24E124136A391163 FunctionBundler result: {"DeduplicationResult":null,"AdrResult":{"NbRepetition":null,"TxPower":null,"DataRate":0,"CanConfirmToDevice":false,"FCntDown":null,"NumberOfFrames":20},"NextFCntDown":102,"PreferredGatewayResult":null}
24E124136A391163 down frame counter: 102
24E124136A391163 valid frame counter, msg: 98 server: 97
24E124136A391163 decoding with: http://decoder/api/DecodeUrsaLinkTemperatureHumidity port: 85
24E124136A391163 sending message {"time":19140303542221364,"tmms":0,"freq":904.1,"chan":0,"rfch":0,"modu":"LoRa","datr":"SF10BW125","rssi":-46.0,"lsnr":13.25,"data":{"temperature":22.6,"humidity":50.0},"port":85,"fcnt":98,"edgets":1642739260499,"rawdata":"A2fiAARoZAYAAQ==","eui":"24E124136A391163","gatewayid":"azure-lora-networkserver"} to hub
24E124136A391163 message '{"temperature":22.6,"humidity":50.0}' sent to hub
24E124136A391163 checking cloud to device message for 00:00:00.8779417
24E124136A391163 done checking cloud to device message, found no message
24E124136A391163 using standard region RX2 frequency 923300000
24E124136A391163 using standard region RX2 datarate DR8
24E124136A391163 using standard region RX2 datarate DR8
24E124136A391163 using standard region RX2 frequency 923300000
24E124136A391163 UnconfirmedDataDown {"DevEui":{"IsValid":true,"AsUInt64":2657444920930406755},"LnsRxDelay":0,"DeviceClassType":0,"Xtime":19140303542221364,"AntennaPreference":0,"Rx1":null,"Rx2":{"DataRate":8,"Frequency":{"AsUInt64":923300000,"InKilo":923300.0,"InMega":923.3,"InGiga":0.9233}},"Data":{"Length":12,"IsEmpty":false},"StationEui":{"IsValid":true,"AsUInt64":2657445937037762055}}
24E124136A391163 UnconfirmedDataDown {"DevEui":{"IsValid":true,"AsUInt64":2657444920930406755},"LnsRxDelay":0,"DeviceClassType":0,"Xtime":19140303542221364,"AntennaPreference":0,"Rx1":null,"Rx2":{"DataRate":8,"Frequency":{"AsUInt64":923300000,"InKilo":923300.0,"InMega":923.3,"InGiga":0.9233}},"Data":{"Length":12,"IsEmpty":false},"StationEui":{"IsValid":true,"AsUInt64":2657445937037762055}}
24E124136A391163 sending message to station with EUI '24E124FFFEF0DE07' with diid -872154575. Payload '604BF9B602A066000AF76A64'.
24E124136A391163 processing time: 00:00:01.5008506
```

A big thanks to the team for getting this issue resolved so quickly!  If you would like me to get the join request,  I can find it in the logs tomorrow.

P.S.  I do that several log messages are duplicated.  It would be nice to get that resolved if possible.

bastbu, 2022-01-21:
Thank you for validating! There's no need to get the logs for the join as we can see the downstream message. I extracted the duplicate logs to a separate issue #1344.

closed 2022-01-21 by bastbu