Azure/iotedge-lorawan-starterkit #1357: Version 2.0.0 - Add StationEui to the message sent to Azire IoT Hub

gerfen, 2022-01-21:
**Is your feature request related to a problem? Please describe.**
With the current basics station implementation, I have no  way of knowing which basics station sent the data.  With the 1.0.x implementation, I have a one-to-one relationship between the packet forwarder and the gateway so I know where the data came from.  I use the gateway name to determine where to send messages to my web app via SignalR.  

With the current 2.0.0 implementation, if I have a single network server servicing many basics stations I lose that level of granularity and my app is broken.

**Describe the solution you'd like**
Add the StationEui to the message sent to Azure IoT hub

```
{"time":19140349910358609,"tmms":0,"freq":905.3,"chan":0,"rfch":0,"modu":"LoRa","datr":"SF10BW125","rssi":-49.0,"lsnr":13.25,"data":{"temperature":20.9,"humidity":61.0},"port":85,"fcnt":868,"edgets":1642785628693,"rawdata":"A2fRAARoegYAAQ==","eui":"24E124136A391163","gatewayid":"azure-lora-networkserver", stationeui: xxxxxxxxxxxxxxxx}
```

gerfen, 2022-01-21:
Looking at the code, this should be straight forward by modifying LoRaWan.NetworkServer.LoraDeviceTelemetry.cs

Add a property
```
 [JsonProperty("stationeui")]
 public string StationEui { get; set; }
```

then modify the constructor

```
 public LoRaDeviceTelemetry(LoRaRequest request, LoRaPayloadData upstreamPayload, object payloadData, byte[] decryptedPayloadData)
        {
            if (request is null) throw new ArgumentNullException(nameof(request));
            var radioMetadata = request.RadioMetadata;
            if (radioMetadata is null) throw new ArgumentException(nameof(radioMetadata));
            if (upstreamPayload is null) throw new ArgumentNullException(nameof(upstreamPayload));
            var datr = request.Region.GetDatarateFromIndex(request.RadioMetadata.DataRate);
            Data = payloadData;
            Rawdata = decryptedPayloadData?.Length > 0 ? Convert.ToBase64String(decryptedPayloadData) : string.Empty;
            Fcnt = upstreamPayload.Fcnt;
            Port = upstreamPayload.Fport.Value;
            Freq = radioMetadata.Frequency.InMega;
            Datr = datr.ToString();
            Rssi = radioMetadata.UpInfo.ReceivedSignalStrengthIndication;
            Rfch = radioMetadata.UpInfo.AntennaPreference;
            Lsnr = radioMetadata.UpInfo.SignalNoiseRatio;
            Time = unchecked(radioMetadata.UpInfo.Xtime); // This is used by former computation only. 
            Chan = (uint)radioMetadata.DataRate; // This is not used in any computation. It is only reported in the device telemetry.
            GpsTime = radioMetadata.UpInfo.GpsTime; // This is not used in any computation. It is only reported in the device telemetry.
            Modu = datr.ModulationKind.ToString(); // This is only used in test path by legacy PacketForwarder code. Safe to eventually remove. Could be also "FSK"
            StationEui = request.StationEui.ToString();
        }
```

gerfen, 2022-01-21:
I did a quick spike to confirm that my proposal will work but I do not know if there our edge cases to consider.  Here are the logs:

```
 sending message {"time":11540474078078575,"tmms":0,"freq":904.5,"chan":0,"rfch":0,"modu":"LoRa","datr":"SF10BW125","rssi":-49.0,"lsnr":13.0,"data":{"temperature":20.7,"humidity":61.5},"port":85,"fcnt":930,"edgets":1642789378878,"rawdata":"A2fPAARoewYAAQ==","eui":"24E124136A391163","gatewayid":"azure-lora-networkserver","stationeui":"24E124FFFEF0DE07","dupmsg":true} to hub
```

Here is the entire join request including the line form above:
```
24E124FFFEF0DE07 Received 'updf' message: '{"msgtype":"updf","MHdr":128,"DevAddr":45545803,"FCtrl":128,"FCnt":930,"FOpts":"","FPort":85,"FRMPayload":"24C3CE07B3D1EC84550C","MIC":30893392,"RefTime":0.000000,"DR":0,"Freq":905100000,"upinfo":{"rctx":0,"xtime":11540474072738935,"gpstime":0,"fts":-1,"rssi":-48,"snr":12.5,"rxtime":1642789373.295493}}'.
02B6F94B querying the registry for device
02B6F94B querying the registry for devices by devAddr 02B6F94B found 1 devices
24E124136A391163 using iotHub directly, no edgeHub queue
24E124136A391163 getting device twin
24E124136A391163 done getting device twin
24E124136A391163 Device registered in cache.
24E124136A391163 converted 16bit FCnt 930 to 32bit FCnt 930
24E124136A391163 no Deduplication Strategy selected
24E124136A391163 FunctionBundler request finished preparing.
24E124136A391163 FunctionBundler request finished preparing.
24E124136A391163 FunctionBundler request: {"GatewayId":"azure-lora-networkserver","ClientFCntUp":930,"ClientFCntDown":940,"Rssi":-48,"AdrRequest":{"DataRate":0,"RequiredSnr":-15,"PerformADRCalculation":false,"FCntUp":930,"FCntDown":940,"MinTxPowerIndex":13,"GatewayId":"azure-lora-networkserver","ClearCache":false},"FunctionItems":5}
24E124136A391163 FunctionBundler result: {"DeduplicationResult":null,"AdrResult":{"NbRepetition":null,"TxPower":null,"DataRate":0,"CanConfirmToDevice":false,"FCntDown":null,"NumberOfFrames":20},"NextFCntDown":946,"PreferredGatewayResult":null}
24E124136A391163 down frame counter: 946
24E124136A391163 valid frame counter, msg: 930 server: 923
24E124136A391163 decoding with: http://decoder/api/DecodeUrsaLinkTemperatureHumidity port: 85
24E124136A391163 sending message {"time":11540474072738935,"tmms":0,"freq":905.1,"chan":0,"rfch":0,"modu":"LoRa","datr":"SF10BW125","rssi":-48.0,"lsnr":12.5,"data":{"temperature":20.7,"humidity":61.5},"port":85,"fcnt":930,"edgets":1642789373547,"rawdata":"A2fPAARoewYAAQ==","eui":"24E124136A391163","gatewayid":"azure-lora-networkserver","stationeui":"24E124FFFEF0DE07"} to hub
24E124136A391163 message '{"temperature":20.7,"humidity":61.5}' sent to hub
24E124136A391163 too late for down message (00:00:05.0083820)
24E124136A391163 processing time: 00:00:05.0279614
24E124FFFEF0DE07 Received 'updf' message: '{"msgtype":"updf","MHdr":128,"DevAddr":45545803,"FCtrl":128,"FCnt":930,"FOpts":"","FPort":85,"FRMPayload":"24C3CE07B3D1EC84550C","MIC":30893392,"RefTime":0.000000,"DR":0,"Freq":904500000,"upinfo":{"rctx":0,"xtime":11540474078078575,"gpstime":0,"fts":-1,"rssi":-49,"snr":13,"rxtime":1642789378.643644}}'.
24E124136A391163 converted 16bit FCnt 930 to 32bit FCnt 930
24E124136A391163 Data message received from station 24E124FFFEF0DE07. Marked as DuplicateDueToResubmission because message already encountered.
24E124136A391163 resubmit from confirmed message detected, msg: 930 server: 930
24E124136A391163 no Deduplication Strategy selected
24E124136A391163 FunctionBundler request finished preparing.
24E124136A391163 FunctionBundler request finished preparing.
24E124136A391163 FunctionBundler request: {"GatewayId":"azure-lora-networkserver","ClientFCntUp":930,"ClientFCntDown":946,"Rssi":-49,"AdrRequest":{"DataRate":0,"RequiredSnr":-15,"PerformADRCalculation":false,"FCntUp":930,"FCntDown":946,"MinTxPowerIndex":13,"GatewayId":"azure-lora-networkserver","ClearCache":false},"FunctionItems":5}
24E124136A391163 FunctionBundler result: {"DeduplicationResult":null,"AdrResult":{"NbRepetition":null,"TxPower":null,"DataRate":0,"CanConfirmToDevice":false,"FCntDown":null,"NumberOfFrames":20},"NextFCntDown":947,"PreferredGatewayResult":null}
24E124136A391163 down frame counter: 947
24E124136A391163 decoding with: http://decoder/api/DecodeUrsaLinkTemperatureHumidity port: 85
24E124136A391163 sending message {"time":11540474078078575,"tmms":0,"freq":904.5,"chan":0,"rfch":0,"modu":"LoRa","datr":"SF10BW125","rssi":-49.0,"lsnr":13.0,"data":{"temperature":20.7,"humidity":61.5},"port":85,"fcnt":930,"edgets":1642789378878,"rawdata":"A2fPAARoewYAAQ==","eui":"24E124136A391163","gatewayid":"azure-lora-networkserver","stationeui":"24E124FFFEF0DE07","dupmsg":true} to hub
24E124136A391163 message '{"temperature":20.7,"humidity":61.5}' sent to hub
24E124136A391163 checking cloud to device message for 00:00:00.4040563
24E124136A391163 done checking cloud to device message, found no message
24E124136A391163 using standard region RX2 datarate DR8
24E124136A391163 using standard region RX2 frequency 923300000
24E124136A391163 UnconfirmedDataDown {"DevEui":{"IsValid":true,"AsUInt64":2657444920930406755},"LnsRxDelay":0,"DeviceClassType":0,"Xtime":11540474078078575,"AntennaPreference":0,"Rx1":{"DataRate":10,"Frequency":{"AsUInt64":925100000,"InKilo":925100.0,"InMega":925.1,"InGiga":0.9251}},"Rx2":{"DataRate":8,"Frequency":{"AsUInt64":923300000,"InKilo":923300.0,"InMega":923.3,"InGiga":0.9233}},"Data":{"Length":12,"IsEmpty":false},"StationEui": [... cut ...] lid":true,"AsUInt64":2657445937037762055}}
24E124136A391163 sending message to station with EUI '24E124FFFEF0DE07' with diid -1694157405. Payload '604BF9B602A0B3030023FF16'.
24E124136A391163 processing time: 00:00:00.6959012
24E124FFFEF0DE07 Received 'dntxed' message: '{"msgtype":"dntxed","seqno":-1694157405,"diid":-1694157405,"DR":10,"Freq":925100000,"DevEui":"24-E1-24-13-6A-39-11-63","rctx":0,"xtime":11540474079078575,"txtime":61349.568573,"gpstime":0}'.
```

Thanks for considering my request.

danigian, 2022-01-24:
Closing the issue as PR #1358 was successfully merged

closed 2022-01-24 by danigian