Azure/iotedge-lorawan-starterkit #1378: Configuration Question:  Concentrator failing to start while configuring BasicsStation for an AAEON AIOT-IP68 gateway with US-915 frequencies

gerfen, 2022-01-25:
I not sure if this is a bug or a configuration error on my part?

I'm in the process of trying to get an AAEON AIOT-IP68 gateway configured as a Basics Station.  I've successfully deployed the basics station module to the device via Azure IoT Hub.  I should note that the RESET_PIUN and SPI_DEV values work with the older 1.0 implementation of the kit.

The twin looks like this:

```
{
    "deviceId": "0242ACFFFE120004",
    "etag": "AAAAAAAAAAI=",
    "deviceEtag": "MTE1NzU4NDQy",
    "status": "enabled",
    "statusUpdateTime": "0001-01-01T00:00:00Z",
    "connectionState": "Disconnected",
    "lastActivityTime": "0001-01-01T00:00:00Z",
    "cloudToDeviceMessageCount": 0,
    "authenticationType": "sas",
    "x509Thumbprint": {
        "primaryThumbprint": null,
        "secondaryThumbprint": null
    },
    "modelId": "",
    "version": 3,
    "properties": {
        "desired": {
            "routerConfig": {
                "NetID": [
                    1
                ],
                "JoinEui": [
                    [
                        "0000000000000000",
                        "FFFFFFFFFFFFFFFF"
                    ]
                ],
                "region": "US902",
                "hwspec": "sx1301/1",
                "freq_range": [
                    902000000,
                    928000000
                ],
                "DRs": [
                    [
                        10,
                        125,
                        0
                    ],
                    [
                        9,
                        125,
                        0
                    ],
                    [
                        8,
                        125,
                        0
                    ],
                    [
                        7,
                        125,
                        0
                    ],
                    [
                        8,
                        500,
                        0
                    ],
                    [
                        0,
                        0,
[... 290 lines cut ...]
An exception occurred while processing requests: System.Net.WebSockets.WebSocketException (0x80004005): The remote party closed the WebSocket connection without completing the close handshake.
   at System.Net.WebSockets.ManagedWebSocket.ThrowIfEOFUnexpected(Boolean throwOnPrematureClosure)
   at System.Net.WebSockets.ManagedWebSocket.EnsureBufferContainsAsync(Int32 minimumRequiredBytes, CancellationToken cancellationToken, Boolean throwOnPrematureClosure)
   at System.Runtime.CompilerServices.PoolingAsyncValueTaskMethodBuilder`1.StateMachineBox`1.System.Threading.Tasks.Sources.IValueTaskSource.GetResult(Int16 token)
   at System.Net.WebSockets.ManagedWebSocket.ReceiveAsyncPrivate[TResult](Memory`1 payloadBuffer, CancellationToken cancellationToken)
   at System.Runtime.CompilerServices.PoolingAsyncValueTaskMethodBuilder`1.StateMachineBox`1.System.Threading.Tasks.Sources.IValueTaskSource<TResult>.GetResult(Int16 token)
   at System.Net.WebSockets.ManagedWebSocket.CloseAsyncPrivate(WebSocketCloseStatus closeStatus, String statusDescription, CancellationToken cancellationToken)
   at LoRaWan.NetworkServer.BasicsStation.Processors.LnsProtocolMessageProcessor.ProcessIncomingRequestAsync(HttpContext httpContext, Func`4 handler, CancellationToken cancellationToken) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/Processors/LnsProtocolMessageProcessor.cs:line 96.
Connection id "0HMEVKB4PAP2G", Request id "0HMEVKB4PAP2G:00000002": An unhandled exception was thrown by the application.
Connection id "0HMEVKB4PAP2H" accepted.
Connection id "0HMEVKB4PAP2H" started.
Request starting HTTP/1.1 GET http://20.112.16.171:5000/router-data/0242ACFFFE120004 - -
1 candidate(s) found for the request path '/router-data/0242ACFFFE120004'
Endpoint '/router-data/{routerId:required} HTTP: GET' with route pattern '/router-data/{routerId:required}' is valid for the request path '/router-data/0242ACFFFE120004'
Request matched endpoint '/router-data/{routerId:required} HTTP: GET'
Executing endpoint '/router-data/{routerId:required} HTTP: GET'
WebSocket connection from 72.0.168.220 established
0242ACFFFE120004 Received 'version' message for station '2.0.5(linux/std)'.
Executed endpoint '/router-data/{routerId:required} HTTP: GET'
Request finished HTTP/1.1 GET http://20.112.16.171:5000/router-data/0242ACFFFE120004 - - - 101 - - 51.3453ms
Connection id "0HMEVKB4PAP2H" disconnecting.
Connection id "0HMEVKB4PAP2H" stopped.
Connection id "0HMEVKB4PAP2H" sending FIN because: "The Socket transport's send loop completed gracefully."
Connection id "0HMEVKB4PAP2H" received FIN.
```

danigian, 2022-01-25:
Hi @gerfen,
It seems to me that you are missing some required values in the createOptions section:

    ```json
     "createOptions": {
            "HostConfig": {
                "NetworkMode": "host",
                "Privileged": true
            },
      "NetworkingConfig": {
            "EndpointsConfig": {
                "host": {}
            }
      }
    ```

Could you please double check and let us know?

In addition, please remind that as for previous packet forwarder module, setting SPI SPEED to 2 is only supported for ARM devices (but you should not need this setting)

gerfen, 2022-01-25:
@danigan Feeling a bit thick at the moment... where should I add the createOptions section?

gerfen, 2022-01-25:
@danigian   I added the createOptions to the lorabasicsstation module config, which is what I think you were asking me to do.

The StationbEUI is getting regenerated on me.  How do I fix the EUI?

gerfen, 2022-01-25:
Adding the createOptions resolved the concentrator start failure but I ended up having to create another Device with the generated StationEui.

```
2022-01-25 18:53:26.570 [TCE:INFO] MUXS reconnect backoff 16s (retry 4)
2022-01-25 18:53:42.597 [AIO:XDEB] [3] ws_connecting state=1
2022-01-25 18:53:42.597 [AIO:XDEB] [3] ws_connecting state=2
2022-01-25 18:53:42.597 [TCE:VERB] Connecting to MUXS...
2022-01-25 18:53:43.651 [AIO:XDEB] [3] ws_connecting state=3
2022-01-25 18:53:43.651 [TCE:VERB] Connected to MUXS.
2022-01-25 18:53:43.651 [AIO:XDEB] [3|WS] > {"msgtype":"version","station":"2.0.5(linux/std)","firmware":null,"package":null,"model":"linux","protocol":2,"features":"rmtsh"}
2022-01-25 18:53:46.391 [AIO:XDEB] [3|WS] < {"msgtype":"router_config","NetID":[1],"JoinEui":[[0,18446744073709551615]],"region":"US902","hwspec":"sx1301/1","freq_range":[902000000,928000000],"DRs":[[10,125,0],[9,125,0],[8,125,0],[7,125,0],[8,500,0],[0,0,0],[0,0,0],[0,0,0],[12,500,1],[11,500,1],[10,500,1],[9,500,1],[8,500,1],[7,500,1]],"sx1301_conf":[{"radio_0":{"enable":true,"freq":904300000},"radio_1":{"enable":true,"freq":905000000},"chan_FSK":{"enable":true,"radio":1,"if":300000},"chan_Lora_std":{
2022-01-25 18:53:46.391 [AIO:XDEB] [3|WS] . "enable":true,"radio":0,"if":300000,"bandwidth":500000,"spread_factor":8},"chan_multiSF_0":{"enable":true,"radio":0,"if":-400000},"chan_multiSF_1":{"enable":true,"radio":0,"if":-200000},"chan_multiSF_2":{"enable":true,"radio":0,"if":0},"chan_multiSF_3":{"enable":true,"radio":0,"if":200000},"chan_multiSF_4":{"enable":true,"radio":1,"if":-300000},"chan_multiSF_5":{"enable":true,"radio":1,"if":-100000},"chan_multiSF_6":{"enable":true,"radio":1,"if":100000},"cha
2022-01-25 18:53:46.391 [AIO:XDEB] [3|WS] . n_multiSF_7":{"enable":true,"radio":1,"if":300000}}],"nocca":true,"nodc":true,"nodwell":true}
2022-01-25 18:53:46.391 [RAL:INFO] Lora gateway library version: Version: 5.0.1;
2022-01-25 18:53:46.391 [RAL:VERB] Connecting to device: /dev/spidev1.0
2022-01-25 18:53:46.392 [RAL:DEBU] SX130x txlut table (0 entries)
2022-01-25 18:53:46.392 [RAL:VERB] SX1301 rxrfchain 0: enable=1 freq=904.3MHz rssi_offset=-166.000000 type=2 tx_enable=1 tx_notch_freq=0
2022-01-25 18:53:46.392 [RAL:VERB] SX1301 rxrfchain 1: enable=1 freq=905.0MHz rssi_offset=-166.000000 type=2 tx_enable=0 tx_notch_freq=0
2022-01-25 18:53:46.392 [RAL:VERB] SX1301 ifchain  0: enable=1 rf_chain=0 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2022-01-25 18:53:46.392 [RAL:VERB] SX1301 ifchain  1: enable=1 rf_chain=0 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2022-01-25 18:53:46.392 [RAL:VERB] SX1301 ifchain  2: enable=1 rf_chain=0 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2022-01-25 18:53:46.392 [RAL:VERB] SX1301 ifchain  3: enable=1 rf_chain=0 freq=200000 bandwidth=0 datarate=0 sync_word=0/0
2022-01-25 18:53:46.392 [RAL:VERB] SX1301 ifchain  4: enable=1 rf_chain=1 freq=-300000 bandwidth=0 datarate=0 sync_word=0/0
2022-01-25 18:53:46.392 [RAL:VERB] SX1301 ifchain  5: enable=1 rf_chain=1 freq=-100000 bandwidth=0 datarate=0 sync_word=0/0
2022-01-25 18:53:46.392 [RAL:VERB] SX1301 ifchain  6: enable=1 rf_chain=1 freq=100000 bandwidth=0 datarate=0 sync_word=0/0
2022-01-25 18:53:46.392 [RAL:VERB] SX1301 ifchain  7: enable=1 rf_chain=1 freq=300000 bandwidth=0 datarate=0 sync_word=0/0
2022-01-25 18:53:46.392 [RAL:VERB] SX1301 ifchain  8: enable=1 rf_chain=0 freq=300000 bandwidth=1 datarate=4 sync_word=0/0
2022-01-25 18:53:46.392 [RAL:VERB] SX1301 ifchain  9: enable=1 rf_chain=1 freq=300000 bandwidth=0 datarate=0 sync_word=0/0
2022-01-25 18:53:46.392 [RAL:VERB] SX130x LBT not enabled
2022-01-25 18:53:46.392 [RAL:VERB] Station device: /dev/spidev1.0 (PPS capture disabled)
2022-01-25 18:53:49.487 [RAL:VERB] Concentrator started (3s95ms)
2022-01-25 18:53:49.487 [S2E:INFO] Configuring for region: US902 -- 902.0MHz..928.0MHz
2022-01-25 18:53:49.487 [S2E:VERB]   DR0  SF10/BW125
2022-01-25 18:53:49.487 [S2E:VERB]   DR1  SF9/BW125
2022-01-25 18:53:49.487 [S2E:VERB]   DR2  SF8/BW125
2022-01-25 18:53:49.487 [S2E:VERB]   DR3  SF7/BW125
2022-01-25 18:53:49.487 [S2E:VERB]   DR4  SF8/BW500
2022-01-25 18:53:49.487 [S2E:VERB]   DR5  FSK
2022-01-25 18:53:49.487 [S2E:VERB]   DR6  FSK
2022-01-25 18:53:49.487 [S2E:VERB]   DR7  FSK
2022-01-25 18:53:49.487 [S2E:VERB]   DR8  SF12/BW500 (DN only)
2022-01-25 18:53:49.487 [S2E:VERB]   DR9  SF11/BW500 (DN only)
2022-01-25 18:53:49.487 [S2E:VERB]   DR10 SF10/BW500 (DN only)
2022-01-25 18:53:49.487 [S2E:VERB]   DR11 SF9/BW500 (DN only)
2022-01-25 18:53:49.487 [S2E:VERB]   DR12 SF8/BW500 (DN only)
2022-01-25 18:53:49.487 [S2E:VERB]   DR13 SF7/BW500 (DN only)
2022-01-25 18:53:49.487 [S2E:VERB]   DR14 undefined
2022-01-25 18:53:49.487 [S2E:VERB]   DR15 undefined
2022-01-25 18:53:49.487 [S2E:VERB]   TX power: 30.0 dBm EIRP
2022-01-25 18:53:49.488 [S2E:VERB]   JoinEui list: 10 entries
2022-01-25 18:53:49.488 [S2E:VERB]   NetID filter: 00000000-00000000-00000000-00000002
2022-01-25 18:53:49.488 [S2E:VERB]   Dev/test settings: nocca=1 nodc=1 nodwell=1
2022-01-25 18:54:31.500 [SYN:INFO] MCU/SX130X drift stats: min: +1.0ppm  q50: +20.0ppm  q80: -38.1ppm  max: +70.5ppm - threshold q90: -41.4ppm
2022-01-25 18:54:31.500 [SYN:INFO] Mean MCU drift vs SX130X#0: 0.5ppm
```

I can see the basicsstation connecting to the gateway now....

```
No PFX is set for the server side authentication. No need to trust any certificate.
No certifiate copied, therefore no update-ca-certificates needed.
[... cut ...]
Endpoint '/router-data/{routerId:required} HTTP: GET' with route pattern '/router-data/{routerId:required}' is valid for the request path '/router-data/024299FFFED2F3E4'
Request matched endpoint '/router-data/{routerId:required} HTTP: GET'
Executing endpoint '/router-data/{routerId:required} HTTP: GET'
WebSocket connection from 72.0.168.220 established
024299FFFED2F3E4 Received 'version' message for station '2.0.5(linux/std)'.
024299FFFED2F3E4 using iotHub directly, no edgeHub queue
024299FFFED2F3E4 getting device twin
024299FFFED2F3E4 done getting device twin
```

Where do I set the StationEui so that it is not auto generated?

gerfen, 2022-01-25:
I'm also seeing lots of this in the basics station logs...

```
2022-01-25 19:17:43.172 [SYN:INFO] MCU/SX130X drift stats: min: +0.5ppm  q50: -7.6ppm  q80: -32.8ppm  max: +40.9ppm - threshold q90: +40.5ppm
2022-01-25 19:17:43.172 [SYN:INFO] Mean MCU drift vs SX130X#0: -1.8ppm
2022-01-25 19:17:43.682 [AIO:XDEB] [3|WS] Ignoring incoming WS PONG
2022-01-25 19:17:45.273 [SYN:VERB] Time sync rejected: quality=378 threshold=348
2022-01-25 19:17:47.374 [SYN:VERB] Time sync rejected: quality=367 threshold=348
2022-01-25 19:17:49.475 [SYN:VERB] Time sync rejected: quality=412 threshold=348
2022-01-25 19:17:51.575 [SYN:VERB] Time sync rejected: quality=386 threshold=348
2022-01-25 19:17:53.676 [SYN:VERB] Time sync rejected: quality=444 threshold=348
2022-01-25 19:17:55.777 [SYN:VERB] Time sync rejected: quality=460 threshold=348
2022-01-25 19:17:57.878 [SYN:VERB] Time sync rejected: quality=458 threshold=348
2022-01-25 19:17:59.979 [SYN:VERB] Time sync rejected: quality=410 threshold=348
2022-01-25 19:18:02.079 [SYN:VERB] Time sync rejected: quality=403 threshold=348
2022-01-25 19:18:04.180 [SYN:INFO] Time sync qualities: min=166 q90=458 max=543 (previous q90=348)
2022-01-25 19:18:04.180 [SYN:VERB] Time sync rejected: quality=543 threshold=458
2022-01-25 19:18:11.533 [SYN:VERB] Time sync rejected: quality=467 threshold=458
```
Not sure if that is normal or problematic?

danigian, 2022-01-25:
Hi @gerfen,

there is the possibility to setup a fixed StationEui by specifying a FIXED_STATION_EUI environment variable in the LoRaBasicsStationModule.

In regards to the Time sync issues, please have a look at this answer on the basicsstation repository: https://github.com/lorabasics/basicstation/issues/41#issuecomment-500751882

gerfen, 2022-01-25:
@danigian I've got the StationEUI fixed now.  Thanks!.

Regarding the time sync issues, I'll let the system run to see if things get better.  TBH, reading through that answer still has me somewhat confused.  :)

I'll close and will open a different one for the time sync issue if it continues to be an issue.

closed 2022-01-25 by gerfen