Azure/iotedge-lorawan-starterkit #1390: Inconsistent Join behavior

gerfen, 2022-01-26:
I've been testing out the new basics station functionality and I'm having inconsistent results with join requests.  I have deployed the network Server module to a Linux VM hosted on Azure and I have deployed the LBS module to an AAEON AIOT-IP^* gateway to act as a basics station.  I am trying to connect an Adeuis network tester to the network.

## Expected Behavior

A device should be able to rejoin the network after getting disconnected.

## Current Behavior

After an initial join, devices struggle to get reconnected.  The join requests show the JoinAccept messages at the network server but the devices fail to join repeatedly.

## Steps to Reproduce

1. Set a up a device in Azure IoT hub
2. See the initial join request and see the data get decoded and forwarded to Azure IoT hub
3. Cycle the power on the device
4. Watch the device struggle to rejoin successfully via the network server logs.

## Context (Environment)

### Device (Host) Operating System

Network Server - Ubuntu 20.04
Basics Station - Ubuntu 20.04

### Architecture

AMD64 for both NS and LBS

### Container Operating System

Linux containers.

### LoRaWAN Module Version

#### Docker

Network Server  - 20.10.11+azure-3
LBS -  20.10.7+azure

## Logs

Network Server logs
```
0018B20000020E90 join request received
0018B20000020E90 querying the registry for OTAA device
0018B20000020E90 updating twin
0018B20000020E90 twin updated
0018B20000020E90 the provided RX2DataRate is not valid
0018B20000020E90 Device registered in cache.
0018B20000020E90 previous device devAddr (02723CE7) removed from cache.
0018B20000020E90 using standard region RX2 datarate DR8
0018B20000020E90 using standard region RX2 frequency 923300000
0018B20000020E90 sending message to station with EUI '024299FFFED2F3E4' with diid -2045090691. Payload '2085CC36887C7F531095A1BF1BE0CDE04E'.
0018B20000020E90 processing time: 00:00:00.1638424
0018B20000020E90 JoinAccept {"DevEui":{"IsValid":true,"AsUInt64":6951112510934672},"LnsRxDelay":5,"DeviceClassType":0,"Xtime":6473925132290188,"AntennaPreference":0,"Rx1":{"DataRate":10,"Frequency":{"AsUInt64":924500000,"InKilo":924500.0,"InMega":924.5,"InGiga":0.9245}},"Rx2":{"DataRate":8,"Frequency":{"AsUInt64":923300000,"InKilo":923300.0,"InMega":923.3,"InGiga":0.9233}},"Data":{"Length":17,"IsEmpty":false},"StationEui":{"IsValid":true,"AsUInt64":162861861309707236}}
024299FFFED2F3E4 Received 'dntxed' message: '{"msgtype":"dntxed","seqno":-2045090691,"diid":-2045090691,"DR":10,"Freq":924500000,"DevEui":"00-18-B2-00-00-02-0E-90","rctx":0,"xtime":6473925137290188,"txtime":63000.042657,"gpstime":0}'.
024299FFFED2F3E4 Received 'jreq' message: '{"msgtype":"jreq","MHdr":0,"JoinEui":"00-18-B2-44-41-52-46-32","DevEui":"00-18-B2-00-00-02-0E-90","DevNonce":51541,"MIC":-248480641,"RefTime":0.000000,"DR":0,"Freq":903900000,"upinfo":{"rctx":0,"xtime":6473925146435892,"gpstime":0,"fts":-1,"rssi":-91,"snr":-13.5,"rxtime":1643159310.689387}}'.
0018B20000020E90 join request received
0018B20000020E90 querying the registry for OTAA device
0018B20000020E90 updating twin
0018B20000020E90 twin updated
0018B20000020E90 the provided RX2DataRate is not valid
0018B20000020E90 Device registered in cache.
0018B20000020E90 previous device devAddr (037F1E90) removed from cache.
0018B20000020E90 using standard region RX2 datarate DR8
0018B20000020E90 using standard region RX2 frequency 923300000
0018B20000020E90 sending message to station with EUI '024299FFFED2F3E4' with diid 1848452255. Payload '206EA6EB3A6F9C19E0466CC09DCAC7D516'.
0018B20000020E90 processing time: 00:00:00.1852848
0018B20000020E90 JoinAccept {"DevEui":{"IsValid":true,"AsUInt64":6951112510934672},"LnsRxDelay":5,"DeviceClassType":0,"Xtime":6473925146435892,"AntennaPreference":0,"Rx1":{"DataRate":10,"Frequency":{"AsUInt64":923300000,"InKilo":923300.0,"InMega":923.3,"InGiga":0.9233}},"Rx2":{"DataRate":8,"Frequency":{"AsUInt64":923300000,"InKilo":923300.0,"InMega":923.3,"InGiga":0.9233}},"Data":{"Length":17,"IsEmpty":false},"StationEui":{"IsValid":true,"AsUInt64":162861861309707236}}
024299FFFED2F3E4 Received 'dntxed' message: '{"msgtype":"dntxed","seqno":1848452255,"diid":1848452255,"DR":10,"Freq":923300000,"DevEui":"00-18-B2-00-00-02-0E-90","rctx":0,"xtime":6473925151435892,"txtime":63014.188373,"gpstime":0}'.
024299FFFED2F3E4 Received 'jreq' message: '{"msgtype":"jreq","MHdr":0,"JoinEui":"00-18-B2-44-41-52-46-32","DevEui":"00-18-B2-00-00-02-0E-90","DevNonce":36124,"MIC":-1303681170,"RefTime":0.000000,"DR":0,"Freq":904100000,"upinfo":{"rctx":0,"xtime":6473925174730228,"gpstime":0,"fts":-1,"rssi":-95,"snr":-13.5,"rxtime":1643159338.983837}}'.
0018B20000020E90 join request received
0018B20000020E90 querying the registry for OTAA device
0018B20000020E90 updating twin
0018B20000020E90 twin updated
0018B20000020E90 the provided RX2DataRate is not valid
0018B20000020E90 Device registered in cache.
0018B20000020E90 previous device devAddr (03D1347F) removed from cache.
0018B20000020E90 using standard region RX2 datarate DR8
0018B20000020E90 using standard region RX2 frequency 923300000
0018B20000020E90 sending message to station with EUI '024299FFFED2F3E4' with diid 980368128. Payload '206872BFEA23D5D8E15F083D7A5B8C66B8'.
0018B20000020E90 processing time: 00:00:00.1568555
0018B20000020E90 JoinAccept {"DevEui":{"IsValid":true,"AsUInt64":6951112510934672},"LnsRxDelay":5,"DeviceClassType":0,"Xtime":6473925174730228,"AntennaPreference":0,"Rx1":{"DataRate":10,"Frequency":{"AsUInt64":923900000,"InKilo":923900.0,"InMega":923.9,"InGiga":0.9239}},"Rx2":{"DataRate":8,"Frequency":{"AsUInt64":923300000,"InKilo":923300.0,"InMega":923.3,"InGiga":0.9233}},"Data":{"Length":17,"IsEmpty":false},"StationEui":{"IsValid":true,"AsUInt64":162861861309707236}}
024299FFFED2F3E4 Received 'dntxed' message: '{"msgtype":"dntxed","seqno":980368128,"diid":980368128,"DR":10,"Freq":923900000,"DevEui":"00-18-B2-00-00-02-0E-90","rctx":0,"xtime":6473925179730228,"txtime":63042.482835,"gpstime":0}'.
```

LBS logs
```
2022-01-26 01:21:19.524 [AIO:XDEB] [3|WS] > {"msgtype":"dntxed","seqno":-75815815,"diid":-75815815,"DR":10,"Freq":923900000,"DevEui":"00-18-B2-00-00-02-0E-90","rctx":0,"xtime":6473925915280332,"txtime":63778.034495,"gpstime":0}
2022-01-26 01:21:19.524 [S2E:INFO] TX 18:b200:2:e90 diid=-75815815 [ant#0] - dntxed: 923.9MHz 30.0dBm ant#0(0) DR10 SF10/BW500 frame=20146C2C6FAF0CF673611FD7..B429E899
2022-01-26 01:21:19.601 [S2E:DEBU] Tx done diid=-75815815
2022-01-26 01:21:39.374 [SYN:INFO] MCU/SX130X drift stats: min: +0.5ppm  q50: -33.8ppm  q80: +45.7ppm  max: +73.8ppm - threshold q90: +73.6ppm
2022-01-26 01:21:39.374 [SYN:INFO] Mean MCU drift vs SX130X#0: 8.2ppm
2022-01-26 01:22:00.975 [any:XDEB] RX mod=LORA f=905100000 bw=125 sz=137 dr=2 BE1182ED0EE750D661727ADADF31B12348507792363E0F59848DD311627F172697FC6327F95F3C78415EC44954F3F57CE8858515EE6F0FEB1A8A7C55D7EE023D3780A0CA6864CAB06BB3DBC079D04CD6820F1CD51603A2B76D624E204E3BF8BF6EF81FDF4157A67CF30777A40470200BBD04AAD9753F159F35B7EE1A034AF90C31C2CBD723BA98C308
2022-01-26 01:22:00.975 [any:XDEB] Dropped frame without CRC or with broken CRC

```
## Additional Information

I am using US-915 frequencies.  I have notice that once the device gets into this state it refuses to reconnect.  I have not found a way to work around this issue yet.

gerfen, 2022-01-27:
With the help of @Mandur and @danigian, I've been able to resolve this issue.  In the twins of the LBS, the radio frequencies were set incorrectly to use the TTN US-915 frequencies - sub-band 2.  I use the frequencies from sub-band 1.

The fix was to change the device twin for the LBS from:

```
    "radio_0": {
        "enable": true,
        "freq": 904300000
    },
    "radio_1": {
        "enable": true,
        "freq": 905000000
    },

```

to 

```
    "radio_0": {
        "enable": true,
        "freq": 902700000
    },
    "radio_1": {
        "enable": true,
        "freq": 903400000
    },
```

For anyone following the sub-bands for US-915 frequencies as defined in the table below: 

LoRaWAN Channels, Frequencies & Sub-bands |   |  
-- | -- | --
Frequency bands | Frequency range (MHz) | Channels
Total USA band | 902.3 - 914.9 | 0-63
Uplink sub-bands | Frequency range (MHz) | Channels
Sub-Band 1 | 902.3 - 903.7 | 0-7
Sub-Band 2 | 903.9 - 905.3 | 8-15
Sub-Band 3 | 905.5 - 906.9 | 16-23
Sub-Band 4 | 907.1 - 908.5 | 24-31
Sub-Band 5 | 908.7 - 910.1 | 32-39
Sub-Band 6 | 910.3 - 911.7 | 40-47
Sub-Band 7 | 911.9 - 913.3 | 48-55
Sub-Band 8 | 915.5 - 914.9 | 56-63
Downlink sub-bands | Frequency range (MHz) | Channels
Downlink sub-band | 903 - 914.2 | 64-71

You are my heroes.  : )  

Thanks for slogging through this issue with me!

closed 2022-01-27 by gerfen

danigian, 2022-01-28:
Thanks @gerfen for your tests!

I have created #1421 and #1422 for adjusting our starter kit.