Azure/iotedge-lorawan-starterkit #1454: Create the Application Insight "Workspace base"

kbeaugrand, 2022-02-02:
**Is your feature request related to a problem? Please describe.**
The classic Application insights is deprecated. 
[image]

**Describe the solution you'd like**
On the template, create a Workspace based Application insights instance.

Add the resource property ``WorkspaceResourceId`` on the template: 
```
{
            "type": "Microsoft.Insights/components",
            "kind": "web",
            "name": "[variables('appInsightName')]",
            "apiVersion": "2015-05-01",
            "location": "[resourceGroup().location]",
            "scale": null,
            "properties": {
                "ApplicationId": "[variables('appInsightName')]",
                "WorkspaceResourceId": "[resourceId('Microsoft.OperationalInsights/workspaces/', variables('logAnalyticsName'))]"
            },
            "dependsOn": [
                "[resourceId('Microsoft.OperationalInsights/workspaces/', variables('logAnalyticsName'))]",
            ]
        },
```

**Describe alternatives you've considered**
Manual migration is available on the portal: 

[image]

**Additional context**
See [https://docs.microsoft.com/en-us/azure/azure-monitor/app/create-workspace-resource](https://docs.microsoft.com/en-us/azure/azure-monitor/app/create-workspace-resource)

bastbu, 2022-02-03:
Thanks @kbeaugrand for reporting this. The changes are now in `dev` and will be released with the next version of the release.

closed 2022-02-03 by bastbu

kbeaugrand, 2022-02-03:
Excellent! Thank you @bastbu