Azure/iotedge-lorawan-starterkit #1556: DeferredTasks handling has to be changed to throw an aggregate exception

danigian, 2022-02-25:


danigian, 2022-02-25:
@bastbu, @atifaziz let's discuss this on monday if there's any question

atifaziz, 2022-02-25:
I prototyped what a helper for this could look like, including max concurrency and cancellation support:

```c#
static class Extensions
{
    public static async Task<(T, Task<TResult>)[]>
        WhenAllCompleted<T, TResult>(this IEnumerable<T> source,
                                     Func<T, CancellationToken, Task<TResult>> function,
                                     int maxConcurrency,
                                     CancellationToken cancellationToken)
    {
        ArgumentNullException.ThrowIfNull(source, nameof(source));
        ArgumentNullException.ThrowIfNull(function, nameof(function));
        if (maxConcurrency <= 0) throw new ArgumentOutOfRangeException(nameof(maxConcurrency), maxConcurrency, null);

        using var semaphore = new SemaphoreSlim(maxConcurrency);
        var pendingTaskList = new List<Task<(int, T, Task<TResult>)>>(source is ICollection { Count: var count } ? count : 4);

        var i = 0;
        foreach (var item in source)
            pendingTaskList.Add(RunAsync(i++, item));

        var results = new (T, Task<TResult>)[pendingTaskList.Count];

        while (pendingTaskList.Count > 0)
        {
            var completedTask = await Task.WhenAny(pendingTaskList).ConfigureAwait(false);
            pendingTaskList.Remove(completedTask);
            (i, var item, var task) = await completedTask.ConfigureAwait(false);
            results[i] = (item, task);
        }
        
        return results;
        
        async Task<(int, T, Task<TResult>)> RunAsync(int i, T item)
        {
            try
            {	        
                await semaphore.WaitAsync().ConfigureAwait(false);
                return (i, item, await Task.WhenAny(function(item, cancellationToken)).ConfigureAwait(false));
            }
            finally
            {
                semaphore.Release();
            }
        }
    }
}
```

atifaziz, 2022-02-28:
We could do this more simply with using `Task.WhenAny` on the task returned by `Task.WhenAll` since awaiting on that doesn't throw and we would get access to the aggregate of exceptions collected by the task return from `Task.WhenAll`.