Azure/iotedge-lorawan-starterkit #1642: LNS logs filled with /metrics endpoint access messages

Metal-Mighty, 2022-03-30:
## Expected Behavior
I would like to be able to choose the logging level for the Prometheus `/metrics` endpoint exposed by the LoRaWanNetworkSrvModule

## Current Behavior
LoRaWanNetworkSrvModule logs are filled with messages every time we call the /metrics endpoint which are useless to us. 
We tried setting LOG_LEVEL to 1 since they looked like debug logs to us, but it only removed one or two lines on each access.

## Context (Environment)

### Container Operating System
Linux containers

### LoRaWAN Module Version
2.0.0

## Logs
```Request starting HTTP/1.1 GET https://LoRaWanNetworkSrvModule:5001/metrics - -
1 candidate(s) found for the request path '/metrics'
Request matched endpoint 'Prometheus metrics'
Executing endpoint 'Prometheus metrics'
Executed endpoint 'Prometheus metrics'
Connection id "0HMGHHC0KK13O" completed keep alive response.
Request finished HTTP/1.1 GET https://LoRaWanNetworkSrvModule:5001/metrics - - - 200 - text/plain;+version=0.0.4;+charset=utf-8 2.8187ms
Request starting HTTP/1.1 GET https://LoRaWanNetworkSrvModule:5001/metrics - -
1 candidate(s) found for the request path '/metrics'
Request matched endpoint 'Prometheus metrics'
Executing endpoint 'Prometheus metrics'
Executed endpoint 'Prometheus metrics'
Connection id "0HMGHHC0KK13O" completed keep alive response.
Request finished HTTP/1.1 GET https://LoRaWanNetworkSrvModule:5001/metrics - - - 200 - text/plain;+version=0.0.4;+charset=utf-8 2.7027ms
Request starting HTTP/1.1 GET https://LoRaWanNetworkSrvModule:5001/metrics - -
1 candidate(s) found for the request path '/metrics'
Request matched endpoint 'Prometheus metrics'
Executing endpoint 'Prometheus metrics'
Executed endpoint 'Prometheus metrics'
Connection id "0HMGHHC0KK13O" completed keep alive response.
Request finished HTTP/1.1 GET https://LoRaWanNetworkSrvModule:5001/metrics - - - 200 - text/plain;+version=0.0.4;+charset=utf-8 2.8362ms```
<!-- Don't forget to remove any connection string information! -->

## Additional Information
We still need detailed logs from the LNS for when we receive data from sensors so we don't want to disable logs altogether

danigian, 2022-03-31:
Hi @Metal-Mighty,
thanks for your comment

The LoRaWAN Network Server is built on top of ASP.NET Core and it uses [logging from the framework](https://docs.microsoft.com/en-us/aspnet/core/fundamentals/logging/?view=aspnetcore-6.0)

As you can see from the documentation, you can specify a different log level for different categories of logs.

From what I see, you might have some benefits by specifying the following environment variable "Logging__LogLevel__Microsoft.AspNetCore" to "Warning" (in combination with the current LOG_LEVEL set to 1)

This should exclude all the verbose logging from AspNetCore routing middleware.

Please let me know if this helps you

Metal-Mighty, 2022-03-31:
Hi Daniele ! Thank you for this tip. 
I tried to clone my deployment to try it, but unfortunately dots aren't allowed in environment variables in the Azure UI, and more globally in Linux environments.
[image]

It has been discussed in [this issue ](https://github.com/dotnet/AspNetCore.Docs/issues/17378) but it's been closed without any hint about how to handle this case.

danigian, 2022-03-31:
I see, what about just setting Logging__LogLevel__Microsoft to Warning? Is it an option for you?

danigian, 2022-04-07:
@Metal-Mighty did you solve this? Can we close the issue?

Metal-Mighty, 2022-04-07:
Hi ! We did try and it looks good! However we didn't have the opportunity to communicate with sensors in the last few days so I'd like to make sure we still have all the logs from incoming data before closing the issue.

Metal-Mighty, 2022-04-15:
@danigian We've tried today and it looks good ! Thanks for your help 🙂

closed 2022-04-15 by Metal-Mighty