Azure/iotedge-lorawan-starterkit #1730: Add Support for AU915-928 Frequency plan

kbeaugrand, 2022-06-14:
**Is your feature request related to a problem? Please describe.**
My customer wants to deploy this solution to Brazil. 
The Frequency plan is ``AU915-928`` and is not yet supported by this starter kit.

**Describe the solution you'd like**
Add the support for AU915-928 Frequency plan.

**Describe alternatives you've considered**
We are using separated LoRaWAN concentrator that support this frequency plan.
Can we jus add the router config for AU915-928 frequency plan to the Concentrator twin ?

**Additional context**
N/A

Mandur, 2022-06-15:
Hello @kbeaugrand,

Thank you for the feature request. As discussed orally, we will try to have a spike branch on this feature soon supporting only basic functionality (upstream messages). Formal integration with the Starterkit will be harder as we do not have the hardware required to test physically those frequency ranges. 

If you have some option where we could run some end to end test in places that supports this frequency plan that would tremendously help with the quality and implementation we can deliver.

kbeaugrand, 2022-06-16:
@Mandur, I asked that to the customer's team.

You can follow the discussion at : [https://github.com/CGI-FR/IoT-Hub-Portal/issues/810#issuecomment-1157545010](https://github.com/CGI-FR/IoT-Hub-Portal/issues/810#issuecomment-1157545010)

Mandur, 2022-09-29:
Hello @Kbeaugrand,

We have a poc of AU implementation on this [feature branch](https://github.com/Azure/iotedge-lorawan-starterkit/tree/feature/add-au-2). Based on our tests, it looks it works for upstream/downstream and join requests. Please keep in mind, **this is PoC state and do not meet the quality of the rest of the kit**. If you find issues please let us know!