Azure/iotedge-lorawan-starterkit #1731: LBS disconnected because of unrecognized LnsMessageType

adcoly, 2022-06-14:
Describe the bug
Everytime my LBS send a discover message and connect to the LNS, it is disconnected due to a message unrecognized by the LNS

## Context (Environment)

###Windows Host OS:
Edition: Windows Server 2019 Datacenter
Version: 17763.2458
Virtual Machine: VM Ware virtual machine

### Architecture
x64

### Container Operating System
Linux containers

### LoRaWAN Module Version
LoRaWANNetworkSrvModule 2.0.0

#### Docker
19.03.15+azure

##Logs
LoraWanNetworkSrvModule logs (debug level):
WebSocket connection from <IP_Adsress> established <GatewayId> Received 'version' message for station '2.0.3(brwodu/debugn)' with package '1.0.0'. <GatewayId> using edgeHub local queue <GatewayId> updating twin <GatewayId> twin updated <GatewayId> using edgeHub local queue <GatewayId> getting device twin <GatewayId> done getting device twin An exception occurred while processing requests: System.FormatException: Could not parse alarm as a valid LnsMessageType. at LoRaWan.NetworkServer.BasicsStation.LnsMessageTypeParser.ParseAndValidate(String s, Nullable1 expectedType) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/LnsMessageTypeParser.cs:line 29
at LoRaWan.NetworkServer.BasicsStation.JsonHandlers.LnsData.<>c__DisplayClass1_0.b__0(String t) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/JsonHandlers/LnsData.cs:line 17
at LoRaWan.NetworkServer.JsonReader.<>c__DisplayClass17_02.<Select>b__0(Utf8JsonReader& rdr) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/JsonReader.cs:line 317 at LoRaWan.NetworkServer.JsonReader.DelegatingJsonReader1.Read(Utf8JsonReader& reader) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/JsonReader.cs:line 328
at LoRaWan.NetworkServer.JsonReader.g__ReadPropertyValue|14_1[T1,T2,T3,T4,T5,T6,T7,T8,T9,T10,T11,T12,T13,T14,T15,T16,TResult,TValue](IJsonProperty1 property, Utf8JsonReader& reader, ValueTuple2& value) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/JsonReader.cs:line 231
at LoRaWan.NetworkServer.JsonReader.<>c__DisplayClass14_017.<Object>b__0(Utf8JsonReader& reader) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/JsonReader.cs:line 197 at LoRaWan.NetworkServer.JsonReader.DelegatingJsonReader1.Read(Utf8JsonReader& reader) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/JsonReader.cs:line 328
at LoRaWan.NetworkServer.JsonReader.Read[T](IJsonReader1 reader, ReadOnlySpan1 utf8JsonTextBytes) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/JsonReader.cs:line 36
at LoRaWan.NetworkServer.JsonReader.Read[T](IJsonReader1 reader, String json) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/JsonReader.cs:line 29 at LoRaWan.NetworkServer.BasicsStation.Processors.LnsProtocolMessageProcessor.HandleDataMessageAsync(StationEui stationEui, IWebSocketWriterHandle1 socket, String json, CancellationToken cancellationToken) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/Processors/LnsProtocolMessageProcessor.cs:line 190
at LoRaWan.NetworkServer.BasicsStation.Processors.LnsProtocolMessageProcessor.InternalHandleDataAsync(RouteValueDictionary routeValues, WebSocket socket, CancellationToken cancellationToken) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/Processors/LnsProtocolMessageProcessor.cs:line 166
at LoRaWan.NetworkServer.BasicsStation.Processors.LnsProtocolMessageProcessor.InternalHandleDataAsync(RouteValueDictionary routeValues, WebSocket socket, CancellationToken cancellationToken) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/Processors/LnsProtocolMessageProcessor.cs:line 177
at LoRaWan.NetworkServer.BasicsStation.Processors.LnsProtocolMessageProcessor.ProcessIncomingRequestAsync(HttpContext httpContext, Func4 handler, CancellationToken cancellationToken) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/Processors/LnsProtocolMessageProcessor.cs:line 77. Connection id "0HMIASQ3N0JEO", Request id "0HMIASQ3N0JEO:00000002": An unhandled exception was thrown by the application. Executed endpoint '/router-data/{routerId:required} HTTP: GET' Request finished HTTP/1.1 GET https://.../router-data/... - - - 101 - - 5846.7286ms Connection id "0HMIASQ3N0JEO" disconnecting.

Mandur, 2022-06-15:
Hello @Adcoly,

Thank you for filing a bug. 
Does this behavior always happen with a particular concentrator (or a particular region?), or are you seeing it for all of them?
I will discuss with some colleagues to find how we can best troubleshoot this

atifaziz, 2022-06-15:
We don't have support for LNS messages where `msgtype` is `alarm` because it's undocumented (in the [LNS protocol specification](https://lora-developers.semtech.com/build/software/lora-basics/lora-basics-for-gateways/?url=tcproto.html)) and considered an experimental feature of the Basics Station [as pointed out in the comment](https://github.com/lorabasics/basicstation/issues/105#issuecomment-878389400) in lorabasics/basicstation#105:

> Basic Station already ships with such a feature. Have a look at [gps.c](https://github.com/lorabasics/basicstation/blob/master/src-linux/gps.c). If you specify the `gps` field in station.conf with a path to the GPS serial device (or a fifo / named pipe), Basic Station will start consuming NMEA (or ubx) sentences and report position changes as `msgtype:alarm` messages to the LNS.
>
> Note, that this feature is considered an _experimental convenience_ function for the following reason:
>
> …

adcoly, 2022-06-15:
I was seeing this behavior only from one gateway, but in fact it was the only one where we were keeping a GPS serial device for the "gps" field in the station.conf.
Thank you both for you answer !
We edited our configuration file and for the moment I did not encounter this error anymore !

closed 2022-06-15 by adcoly