Azure/iotedge-lorawan-starterkit #1884: [Story] Migrate Infrastructure to Bicep

Mandur, 2022-10-11:
## Description

We want to migrate the LWSK to use bicep as IaC language in order to benefit from the latest language features and remove unneeded endpoint.

## Acceptance criteria

- [ ] The [existing IaC code](https://github.com/Azure/iotedge-lorawan-starterkit/tree/dev/Template) is migrated to Bicep and committed in bicep in the dev branch
- [ ] The new infrastructure generation code is tested manually
- [ ] [Stretch] The Azure Function [createEdgeDevice](https://github.com/Azure/iotedge-lorawan-starterkit/blob/dev/LoRaEngine/LoraKeysManagerFacade/CreateEdgeDevice.cs) is decommissioned and its logic run in the Bicep

## Resources

- [Spike done at OCW22 to experiment with bicep](https://github.com/Azure/iotedge-lorawan-starterkit/tree/feature/add-bicep-infra/Template)

kbeaugrand, 2022-10-19:
Hi, question on that topic, will the bicep template be stored in a public registry ?

Mandur, 2022-10-19:
Hello @kbeaugrand,

This is currently not being planned but out of curiosity why would it interest you? Is there something with the portal?

kbeaugrand, 2022-10-19:
Yes, we are thinking about providing bicep templates for the portal. But we'd like to still use the starterkit template dependency. 
But it seems that template dependency inside bicep needs to use templates that are published in a registry...

Mandur, 2022-11-01:
As discussed the compiled bicep will be published from now on as part of our releases

closed 2022-11-01 by Mandur