Azure/iotedge-lorawan-starterkit #1916: Remove the Facade auth code from the logs

kbeaugrand, 2022-10-25:
## Expected Behavior

Credentials should be obfuscated event in logs.

## Current Behavior

On the ``LoRaWANNetworkSrvModule`` we can see in clear the Facade auth code

[image]

## Steps to Reproduce

1. Go the the LoRaWANNetworkSrvModule logs

## Context (Environment)

### Device (Host) Operating System

N/A 

### Architecture

N/A

### Container Operating System

Linux containers 

### LoRaWAN Module Version

2.2.0

#### Docker

N/A

## Logs

## Additional Information

Mandur, 2022-10-26:
Hello @kbeaugrand 

Thank you for reporting this, we will have a look

Mandur, 2022-10-26:
By adding the environment variable 
        "Logging:LogLevel:System": "None"
should disable this behavior, we will move into integrating this in upstream.

closed 2022-11-01 by Mandur

kbeaugrand, 2022-11-07:
Fine, that works, but we do not have any logs about query executed on the Facade. 
Not a big deal