Azure/iotedge-lorawan-starterkit #1978: DRs set to 0 in US915 router config recognized as FSK instead of "undefined"

Metal-Mighty, 2022-11-08:
## Expected Behavior

In the US915 config, DR 5-7 are supposed to be undefined, as specified by the LoRa Alliance. 

[image]

## Current Behavior

In our US915 configuration, undefined DRs are set to `[0, 0, 0]` in order to ignore them in our Basic Station. But during our tests, we noticed that our Basic Station recognizes DRs 5-7 as FSK and not as undefined like the DRs 14-15. From what I could see in the LNS code, if the value is set to 0 then it implies that the value is to be considered as FSK and not a "classic" DR. This is compliant to another definition in the LoRa developer portal.

[image]

Since we are currently having big LoRa signal issues in the US (and only this area, no issue in Europe or China), we are wondering if that could be a cause for it.

We have been in contact with Cisco (the manufacturer for our hardware) and they told us that they use `-1` as the first value for unused DRs (i.e. `[-1, 0, 0]`). This is however not possible in the LoRaWanNetworkSrvModule  since it only accepts 0 or values 7-12.

A screenshot of our LBS logs and the content of our router_config are provided below. We are using the FSB 02 of the US915 frequency plan.

## Steps to Reproduce

1.
2.
3.
4.

## Context (Environment)

### Device (Host) Operating System

Windows Server 2019 with EFlow

### Architecture

x64

### Container Operating System

EFlow

### LoRaWAN Module Version
2.1.0

#### Docker

## Logs

[image]

## Additional Information

```json
"routerConfig": {
      "NetID": [
          1
      ],
      "JoinEui": [
          [
              "0000000000000000",
              "FFFFFFFFFFFFFFFF"
          ]
      ],
      "region": "US902",
      "hwspec": "sx1301/1",
      "freq_range": [
          902000000,
          928000000
      ],
      "DRs": [
          [
              10,
              125,
              0
          ],
          [
              9,
              125,
              0
          ],
          [
              8,
              125,
              0
          ],
          [
              7,
              125,
              0
          ],
          [
              8,
              500,
              0
          ],
          [
              0,
              0,
              0
          ],
          [
              0,
              0,
              0
          ],
          [
              0,
              0,
              0
          ],
          [
              12,
              500,
              1
          ],
          [
              11,
              500,
              1
          ],
          [
              10,
              500,
              1
          ],
          [
              9,
              500,
              1
          ],
          [
              8,
              500,
              1
          ],
          [
              7,
              500,
              1
          ]
      ],
      "sx1301_conf": [
          {
              "radio_0": {
                  "enable": true,
                  "freq": 904300000,
                  "radio": 0,
                  "if": 0,
                  "bandwidth": 0,
                  "spread_factor": 0
              },
              "radio_1": {
                  "enable": true,
                  "freq": 905000000,
                  "radio": 0,
                  "if": 0,
                  "bandwidth": 0,
                  "spread_factor": 0
              },
              "chan_FSK": {
                  "enable": false,
                  "freq": 0,
                  "radio": 1,
                  "if": 300000,
                  "bandwidth": 0,
                  "spread_factor": 0
              },
              "chan_Lora_std": {
                  "enable": true,
                  "freq": 0,
                  "radio": 0,
                  "if": 300000,
                  "bandwidth": 500000,
                  "spread_factor": 8
              },
              "chan_multiSF_0": {
                  "enable": true,
                  "freq": 0,
                  "radio": 0,
                  "if": -400000,
                  "bandwidth": 0,
                  "spread_factor": 0
              },
              "chan_multiSF_1": {
                  "enable": true,
                  "freq": 0,
                  "radio": 0,
                  "if": -200000,
                  "bandwidth": 0,
                  "spread_factor": 0
              },
              "chan_multiSF_2": {
                  "enable": true,
                  "freq": 0,
                  "radio": 0,
                  "if": 0,
                  "bandwidth": 0,
                  "spread_factor": 0
              },
              "chan_multiSF_3": {
                  "enable": true,
                  "freq": 0,
                  "radio": 0,
                  "if": 200000,
                  "bandwidth": 0,
                  "spread_factor": 0
              },
              "chan_multiSF_4": {
                  "enable": true,
                  "freq": 0,
                  "radio": 1,
                  "if": -300000,
                  "bandwidth": 0,
                  "spread_factor": 0
              },
              "chan_multiSF_5": {
                  "enable": true,
                  "freq": 0,
                  "radio": 1,
                  "if": -100000,
                  "bandwidth": 0,
                  "spread_factor": 0
              },
              "chan_multiSF_6": {
                  "enable": true,
                  "freq": 0,
                  "radio": 1,
                  "if": 100000,
                  "bandwidth": 0,
                  "spread_factor": 0
              },
              "chan_multiSF_7": {
                  "enable": true,
                  "freq": 0,
                  "radio": 1,
                  "if": 300000,
                  "bandwidth": 0,
                  "spread_factor": 0
              }
          }
      ],
      "nocca": true,
      "nodc": true,
      "nodwell": true
  }

danigian, 2022-11-11:
Hi @Metal-Mighty, thanks for reporting the issue.

I just created #1981 to properly handle the "undefined" SF for skipping the DR setting in scenarios like the US one.

Before it gets merged and released, you could eventually build the docker image from that branch and try it out.
If you decide to do so, please redeploy also the Azure Function and keep in mind that [v2.2.0](https://github.com/Azure/iotedge-lorawan-starterkit/releases/tag/v2.2.0) brought a breaking change when it comes to handling Application Insights connection string.

Metal-Mighty, 2022-11-14:
Thank you @danigian ! We will try to build and run it asap to see if it fixes our issues.
Do you have a roadmap with the date for each next release ?

danigian, 2022-11-15:
Thanks for trying the fix out @Metal-Mighty, we currently have an issue with our local CI environment which we are hopefully going to fix soon. No fixed date for the next releases, but if the fix works for you and no other issues rise, as soon as the CI is fixed, we could release a new version.

adcoly, 2022-11-16:
We manage to deploy this new LNS image and we have an issue when trying to recognize our Gateway :

```
Connection id "0HMM7NLC89MII", Request id "0HMM7NLC89MII:00000002": An unhandled exception was thrown by the application.
<Gateway Dev EUI> Received discovery request from: <Gateway Dev EUI>
<Gateway Dev EUI> Received 'version' message for station '2.0.3(brwodu/debugn)' with package '1.0.0'.
An exception occurred while processing requests: LoRaWan.LoRaProcessingException: The configuration request of station '<Gateway Dev EUI>' did not match any configuration in IoT Hub. If you expect this connection request to succeed, make sure to provision the Basics Station in the device registry.
ErrorCode: InvalidDeviceConfiguration
   at LoRaWan.NetworkServer.BasicsStation.BasicsStationConfigurationService.SetReportedPackageVersionAsync(StationEui stationEui, String package, CancellationToken cancellationToken) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/BasicsStationConfigurationService.cs:line 141
   at LoRaWan.NetworkServer.BasicsStation.Processors.LnsProtocolMessageProcessor.HandleDataMessageAsync(StationEui stationEui, IWebSocketWriterHandle`1 socket, String json, CancellationToken cancellationToken) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/Processors/LnsProtocolMessageProcessor.cs:line 139
   at LoRaWan.NetworkServer.BasicsStation.Processors.LnsProtocolMessageProcessor.InternalHandleDataAsync(RouteValueDictionary routeValues, WebSocket socket, CancellationToken cancellationToken) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/Processors/LnsProtocolMessageProcessor.cs:line 108
   at LoRaWan.NetworkServer.BasicsStation.Processors.LnsProtocolMessageProcessor.InternalHandleDataAsync(RouteValueDictionary routeValues, WebSocket socket, CancellationToken cancellationToken) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/Processors/LnsProtocolMessageProcessor.cs:line 119
   at LoRaTools.WebSocketConnection.HandleAsync(Func`4 handler, CancellationToken cancellationToken) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoraTools/WebSocketConnection.cs:line 41
   at LoRaWan.NetworkServer.BasicsStation.Processors.LnsProtocolMessageProcessor.<>c__DisplayClass14_0.<<HandleDataAsync>b__0>d.MoveNext() in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/Processors/LnsProtocolMessageProcessor.cs:line 85
--- End of stack trace from previous location ---
   at LoRaWan.NetworkServer.BasicsStation.Processors.LnsProtocolMessageProcessor.ExecuteWithExceptionHandlingAsync[T](Func`1 action) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/Processors/LnsProtocolMessageProcessor.cs:line 222
.
```

danigian, 2022-11-16:
Hi @adcoly, did you also redeploy the new Azure Functions version?

adcoly, 2022-11-16:
Yes the Function App was updated with the v2.2.1.
We have an Application Insight linked to our Function App and no issue/error are raised through it. So I can't see any error from the Function App.

danigian, 2022-11-16:
@adcoly, navigating the code that can lead to the error message you are reporting, the only possibility is that LNS is not able to properly get the primary key for the BasicStation device in IoT Hub (and this is happening via communication with the Azure Function). There must be something going off there. Please double check settings of environment variables / edge twin module for the LoRaWanNetworkSrvModule

adcoly, 2022-11-17:
Ok, i'll start by verifying that !

I also would like to provide the version of each part of the installation to be sure you don't see any incompabilities in there :
- EFLOW : LTS 1.1.2202.21022
- IoT Edge : 1.1
- edgeHub & edgeAgent : 1.4
- Function App : 2.2.1
- LNS : the fix provided

danigian, 2022-11-18:
@adcoly I just realized there is unfortunately another bug (#1984) that creates a wrong zip for the Azure Functions. Can you please redeploy the Azure Functions now? I have created the zip file with the proper structure.