Azure/iotedge-lorawan-starterkit #2042: QUESTION:  Azure IoT TLS upgrade

gerfen, 2023-02-01:
I'm looking for some guidance on whether this [notice](https://techcommunity.microsoft.com/t5/internet-of-things-blog/azure-iot-tls-critical-changes-are-almost-here-and-why-you/ba-p/2393169) regarding the change to the Baltimore root certificate will impact my deployment to Azure IoT Hub?  If so, what steps should I take to get into compliance?

Thanks!

danigian, 2023-02-07:
Hi @gerfen, the answer really depends on your setup.
If you are just using the release [v2.2.1](https://github.com/Azure/iotedge-lorawan-starterkit/releases/tag/v2.2.1) of this starter kit, you should be ready to go.

The published version of [loraedge/lorawannetworksrvmodule](https://hub.docker.com/r/loraedge/lorawannetworksrvmodule/tags) is including the DigiCert Global G2 certificate as trusted root, therefore no further action should be required.

In general, the blog post you linked is including a "Validation" section where you find information on how to validate if any certificate related issue could be there or not.

gerfen, 2023-02-07:
@danigian Thank you for your guidance.  I need to update several of my production deployments to the latest version and this has finally pushed me to do so.

closed 2023-02-07 by gerfen

danigian, 2023-02-07:
@gerfen updating is always a great thing 👍 Just be aware that there are some breaking changes, depending on the version you start from.