Azure/iotedge-lorawan-starterkit #2069: Add timestamp/datetime to console logs

Metal-Mighty, 2023-03-29:
**Is your feature request related to a problem? Please describe.**
The lack of timestamps in the LNS logs sometimes makes it hard to debug issues while using Azure's troubleshooting tools or Docker containers logs.

**Describe the solution you'd like**
Pending a possible implementation of log formatters (see #1131), a date/time before the actual log would help a lot. 
I would expect it to be in UTC for consistency across various timezones.

**Describe alternatives you've considered**
-

**Additional context**
-

Mandur, 2023-04-03:
Hello @metal-mighty,

Could you please give some details about the issue requested, in my Azure Log management I am seeing correctly the timecreated of each log line (cf below)
[image]

When using docker logs are you using [the -t option](https://docs.docker.com/engine/reference/commandline/logs/#options)? For me this seems to add the timestamp as expected.
[image]

Mandur, 2023-04-03:
As far as I know Docker generates timestamps attached to each logs, this is how the -t command (and the Azure monitor) is able to get the emission time. The screenshot above is coming from our CI using Azure monitor (I think you are using ELK and not Azure monitor). 

I am not a filebeat specialist, but based on diverse articles, it seems that filebeat do take the docker timestamp by default (source : https://www.elastic.co/guide/en/beats/filebeat/current/filebeat-input-container.html), can you paste your logfile config here to compare?

Are you sure you don't have this timestamp field exposed in your logs? I would recommend asking this question to a filebeat specialist as it is not a technology I am very familiar with.

Metal-Mighty, 2023-04-04:
Ah there might be a misunderstanding: we don't have any issue when publishing the logs in our ELK, but they aren't always deployed in our production environments. 
Each of our clients have to deploy machines to handle our logs and forward them to our main ELK instance, but this deployment is handled by another team and isn't in sync with our deployments. 

That's why we rely a lot on the Troubleshoot screen in the Aure IoT Hub's Edge Devices, even more so when we don't have a direct access to the clients' machines (mainly due to security), and why the lack of timestamps on this particular screen can impact us.