Azure/iotedge-lorawan-starterkit #2099: Could not get Facade information from module twin

Metal-Mighty, 2023-09-13:
## Expected Behavior
Connection of the LNS to the Facade to get module twin information

## Current Behavior
The LNS fails to connect to the Facade and crashes. I provide various information below

## Steps to Reproduce
-

## Context (Environment)

### Device (Host) Operating System
Ubuntu 20.04

### Architecture
x64

### Container Operating System
Linux Containers

### LoRaWAN Module Version
2.2.1

#### Docker
20.10.25+azure-2

## Logs
LoRaWanNetworkSrvModule startup logs:
```No file found at /var/lorastarterkit/certs/client.ca.crt. Nothing was copied over.
No certifiate copied, therefore no update-ca-certificates needed.
Getting properties from module twin...
The initial configuration of the facade function could not be found in the desired properties
LoRaDeviceCache cleared.
LoRaDeviceCache cleared.
Unhandled exception. System.Configuration.ConfigurationErrorsException: Could not get Facade information from module twin
   at LoRaWan.NetworkServer.BasicsStation.ModuleConnection.ModuleConnectionHost.InitModuleAsync(CancellationToken cancellationToken) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/ModuleConnection/ModuleConnectionHost.cs:line 78
   at LoRaWan.NetworkServer.BasicsStation.ModuleConnection.ModuleConnectionHost.CreateAsync(CancellationToken cancellationToken) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/ModuleConnection/ModuleConnectionHost.cs:line 49
   at LoRaWan.NetworkServer.BasicsStation.BasicsStationNetworkServer.RunServerAsync(NetworkServerConfiguration configuration, CancellationToken cancellationToken) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/BasicsStationNetworkServer.cs:line 59
   at LoRaWan.NetworkServer.BasicsStation.BasicsStationNetworkServer.RunServerAsync(NetworkServerConfiguration configuration, CancellationToken cancellationToken) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWan.NetworkServer/BasicsStation/BasicsStationNetworkServer.cs:line 67
   at Program.<Main>$(String[] args) in /build/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoRaWanNetworkSrvModule/Program.cs:line 35
   at Program.<Main>(String[] args)
./start.sh: line 38:     8 Aborted                 (core dumped) dotnet LoRaWanNetworkSrvModule.dll
```

LNS Container Create Options:
```
{
  "ExposedPorts": {
    "5000/tcp": {},
    "5001/tcp": {}
  },
  "HostConfig": {
    "Binds": [
      "/var/opt/ssl/network-server.pfx:/usr/share/local/ssl/network-server.pfx:ro"
    ],
    "PortBindings": {
      "5000/tcp": [
        {
          "HostPort": "5000"
        }
      ],
      "5001/tcp": [
        {
          "HostPort": "5001"
        }
      ]
    }
  }
}
```

LNS Module Twin: 
```json
{
    "etag": "<etag>",
    "deviceId": "<device id>",
    "deviceEtag": "<deviceEtag>",
    "moduleId": "LoRaWanNetworkSrvModule",
    "version": 4,
    "tags": {},
    "properties": {
        "desired": {
            "$metadata": {
                "$lastUpdated": "2023-09-13T07:37:49.4881485Z",
                "$lastUpdatedVersion": 3,
                "FacadeAuthCode": {
                    "$lastUpdated": "2023-09-13T07:37:49.4881485Z",
                    "$lastUpdatedBy": "<deployment name>-lns-221",
                    "$lastUpdatedByDigest": "638301874678971427",
                    "$lastUpdatedVersion": 3
                },
                "FacadeServerUrl": {
                    "$lastUpdated": "2023-09-13T07:37:49.4881485Z",
                    "$lastUpdatedBy": "<deployment name>-lns-221",
                    "$lastUpdatedByDigest": "638301874678971427",
                    "$lastUpdatedVersion": 3
                }
            },
            "$version": 3,
            "FacadeAuthCode": "<FunctionApp 'default' App Key>",
            "FacadeServerUrl": "https://<FunctionApp resource name>.azurewebsites.net/api/"
        },
        "reported": {
            "$metadata": {
                "$lastUpdated": "0001-01-01T00:00:00.0000000Z"
            },
            "$version": 1
        }
    },
    "configurations": {
        "<deployment name>": {
            "status": "Targeted"
        },
        "<deployment name>-lns-221": {
            "status": "Applied"
        }
    },
    "status": "enabled",
    "statusUpdateTime": "0001-01-01T00:00:00.0000000Z",
    "lastActivityTime": "0001-01-01T00:00:00.0000000Z",
    "connectionState": "Disconnected",
    "cloudToDeviceMessageCount": 0,
    "authenticationType": "sas"
}
```

## Additional Information
* Our deployment is automated and has been used in the past for other resource groups and we haven't noticed changes that should impact the creation of the new resources. 
* The function app is deployed using the same version as the LNS ([v2.2.1](https://github.com/Azure/iotedge-lorawan-starterkit/releases/download/v2.2.1/function-2.2.1.zip))
* We were using v2.1.0 before and updated to 2.2.1 just to be sure, with no change
* We checked all passwords, access keys and URIs and they look correct to us
* The FA connects to IoT Hub with the `iothubowner` connection string so there shouldn't be any rights issue

Mandur, 2023-09-21:
Hello,

Thanks for reporting the bug, we are going to investigate!

Mandur, 2023-11-13:
This was due to a problem with the underlying IoT infrastructure, recreating it fixed the above issue

closed 2023-11-13 by Mandur