Azure/iotedge-lorawan-starterkit #2123: No matching manifest for linux/arm/v7

Terrence77, 2024-01-22:
Azure IoT Edge LoRaWAN Starter Kit

I have installing AzureIotEdge to my Raspberry Pi with the Seeed Stuio Wm1002 lora concentrator.

I noticed in My IOTHub / Devices / MyGateway that LoRaWanNetworkSrvModule and LoRaBasicsStationModule were showing a runtime status as Error.

I went to my gateway and issued this command to look at the logs and found this error.

sudo iotedge system logs

2024-01-17T01:50:03Z [WARN] - Could not pull image loraedge/lorabasicsstationmodule:2.2.2
2024-01-17T01:50:03Z [WARN] - caused by: no matching manifest for linux/arm/v7 in the manifest list entries

No manifest for linux/arm/v7  ??
Doing research I noticed on this website that the linux/arm/v7 2.2.2 file does exist.
https://hub.docker.com/r/loraedge/lorabasicsstationmodule/tags

Ideas on solving this issue?

Thank you.

Terrence77, 2024-01-23:
I know this probably is not the place for tech support, but I have searched the net and can't find any forum answering questions about this great project.

Suggestions?

kbeaugrand, 2024-01-23:
@Terrence77, is this a transient error ? 

As far I know by looking in the manifest, there is a tag for  linux/arm/v7: 
[image]

Otherwise, you can manually specify to download the required image for testing purpose: loraedge/lorabasicsstationmodule:2.2.2-arm32v7

Terrence77, 2024-01-23:
@kbeaugrand Thank you for your reply.

Yes, that is what I was trying to get across in my post, that the file DOES exist.
>>Doing research I noticed on this website that the linux/arm/v7 2.2.2 file does exist.
>>https://hub.docker.com/r/loraedge/lorabasicsstationmodule/tags

>>is this a transient error 
Well I am seeing the error in the Azure portal.
[image]

>>Otherwise, you can manually specify to download the required image for testing purpose: loraedge/lorabasicsstationmodule:2.2.2-arm32v7

I "think" I tried that but was not successful.

I am new to containers so I am sure I mangled my syntax.

Terrence77, 2024-01-24:
@kbeaugrand 

>>Otherwise, you can manually specify to download the required image for testing purpose: loraedge/lorabasicsstationmodule:2.2.2-arm32v7

Can I specify this in the Azure Portal, or do I need to do it in Bash on the gateway?

Thanks.

kbeaugrand, 2024-01-24:
> @kbeaugrand 
> 
> 
> 
> >>Otherwise, you can manually specify to download the required image for testing purpose: loraedge/lorabasicsstationmodule:2.2.2-arm32v7
> 
> 
> 
> Can I specify this in the Azure Portal, or do I need to do it in Bash on the gateway?
> 
> 
> 
> Thanks.

Please share us the configuration you are using in the portal for the module.

Terrence77, 2024-01-24:
Here is the hardware config and setup.

`
Rasp Pi 2B
Pi OS Lite 32bit Debian Bulllseye
Seeed WM1302

sudo raspi-config
Endable SPI interface
Endable IC2 interface

sudo apt update
sudo apt install -y git
cd ~
git clone https://github.com/Lora-net/sx1302_hal
cd ~/sx1302_hal
make

curl https://packages.microsoft.com/config/debian/11/packages-microsoft-prod.deb > ./packages-microsoft-prod.deb
sudo apt-get install ./packages-microsoft-prod.deb
sudo apt-get update
sudo apt-get install moby-engine
sudo curl -L https://github.com/Azure/azure-iotedge/releases/download/1.2.5/aziot-identity-service_1.2.4-1_debian11_armhf.deb -o aziot-identity-service.deb && sudo apt-get install ./aziot-identity-service.deb
sudo curl -L https://github.com/Azure/azure-iotedge/releases/download/1.2.5/aziot-edge_1.2.5-1_debian11_armhf.deb -o aziot-edge.deb && sudo apt-get install ./aziot-edge.deb

pi@raspberrypi:~ $ sudo iotedge system status
System services:
    aziot-edged             Running
    aziot-identityd         Running
    aziot-keyd              Running
    aziot-certd             Running
    aziot-tpmd              Ready

`

Terrence77, 2024-01-24:
[image]

[image]

[image]

[image]

[image]

[image]

[image]

Terrence77, 2024-01-24:
The script that ran when I clicked "Deploy to Azure" setup the devices in my IOTHub.
I note that one of the devices is a concentrator (my Seeed WM1302?).
The device twin shows it being a 1301, NOT a 1302.

[image]

[image]

Mandur, 2024-03-13:
Apologies for not seeing the issue earlier, are you still trying to fix it?

Did you try to run "docker pull loraedge/lorabasicsstationmodule:2.2.2-arm32v7" on your target device?

Mandur, 2024-03-15:
Understood, sorry we did not see the request earlier, have a great weekend!

closed 2024-03-15 by Mandur