Azure/iotedge-lorawan-starterkit #2127: Use NetworkServer Module with external gateway?

roman-ruthofer, 2024-05-13:
**Description**
We are using the following indoor lora gateway
https://www.thermokon.de/direct/en-gb/categories/gateway-lrw-indoor-femto

Is there any way to use the NetworkServer Module with such external gateways? It seems like the gateway is using UDP and the NetworkServer Module only exports TCP Ports.

kbeaugrand, 2024-05-13:
Hi, 

Network server supports Lora Basic station connectivity. 

It seems that regarding the datasheet of your gateway that is also supported. 

I suggest you to look at this side.

roman-ruthofer, 2024-05-14:
> Hi,
> 
> Network server supports Lora Basic station connectivity.
> 
> It seems that regarding the datasheet of your gateway that is also supported.
> 
> I suggest you to look at this side.

I'll have a look. Thanks!

closed 2024-05-15 by roman-ruthofer