Azure/iotedge-lorawan-starterkit #2128: Use LoRaWanNetworkSrvModule v2.xxx on IoT Edge Device with external PackageForwarder

roman-ruthofer, 2024-05-16:
**Description**
We have 
- an external Antenna that can forward Lora Packages to a LNS via UDP.
- an existing IoT Edge Device running Iot Edge Runtime and LoRaWanNetworkSrvModule

With LoRaWanNetworkSrvModule.1.0.7 we could simply foreward the packages from the antenna to the LoRaWanNetworkSrvModule on our IoT Edge Device via UDP and everything worked fine.

How can we achive this with LoRaWanNetworkSrvModule.2.xxx?

Mandur, 2024-05-25:
Hello @roman-ruthofer,

Since the version 2.0.0 we are supporting the [LoRa Basics station](https://doc.sm.tc/station/) protocol. This new protocol uses websocket instead of UDP. Could I ask a bit more about your scernario, are you manually forwarding the packets to the LNS?

roman-ruthofer, 2024-05-27:
Hi @Mandur 

We have the following scenario:
[image]

The Thermokon Gateway is able to act as a Package Forwarder (with UDP) or should be able to work with a LoRa Basics station. https://www.thermokon.de/direct/categories/gateway-lrw-indoor-femto

Our IoT Edge Device Hardware has no radio antenna/hardware on board and we are not able to change that.